autowarefoundation/autoware #1799: [feature] CNN based LiDAR segmentation from Baidu

amc-nu, 2018-12-21:
# New Feature
Integrate and adapt Baidu's CNN LiDAR segmentation into Autoware.

AbangLZU, 2018-12-28:
Is this the caffe model and prototxt you used? 
 https://github.com/ApolloAuto/apollo/tree/master/modules/perception/production/data/perception/lidar/models/cnnseg

I test this node on this 64 pretrained model and KITTI dataset, then catch some error...

AbangLZU, 2019-01-02:
I have solved this problem. It seems like that the pretrained model in the Apollo project only used six channel features (without the direction_data_ and distance_data_ ). I suggest to add a paramter (like use_constant_feature) to set whether to use the direction_data_ and distance_data_ features.

closed 2019-02-18 by amc-nu

4n00815, 2019-05-25:
@AbangLZU can you please tell me how did you solve this issue? I have the same problem. Thank you

zzhh00, 2020-01-16:
@AbangLZU 
Hi,I meet the same problem when runnung apollo cnn segmentation.Can you please tell where you modified to solve this problem?Many thanks.

ishfaq-gist, 2020-03-04:
Hi, you solved this issue? I am facing same issue. Where we need to modified.  Help is requested please. 

Thank you

zzhh00, 2020-03-04:
@ishfaq-gist 
Hi,this issue can be solved as below:
1.download the corresponding pretrained model from this link:
https://github.com/ApolloAuto/apollo/tree/master/modules/perception/production/data/perception/lidar/models/cnnseg 
files deploy.prototxt and deploy.caffemodel are needed.

2.modify the source codes following this link:
https://github.com/autowarefoundation/autoware/pull/1826/commits/cd92761a62435404b0796405fcb4493ee06f9a7a
However the result is really poor,hope u can have some progress.

ishfaq-gist, 2020-03-05:
thank you @zzhh00  .I solved the issue. thank you