autowarefoundation/autoware #1813: Release 1.11

kfunaoka, 2018-12-26:
# Release Schedule

# Announcements

# Security Fixes

# New Features

# Improvements

# Bug Fixes

kfunaoka, 2018-12-27:
I'd like to add #1811 and #1812 for more maintainability.
#1812 can delete a part of obsolete ones (not all of them).

kfunaoka, 2018-12-27:
@mitsudome-r It should be! I've added #1817 into the headline.

I'd like to add #1609. It's too huge and should be well organized. @s-azumi, @mitsudome-r, and me will take care of it.

cirpue49, 2019-01-07:
Hope to add #1772 #1461

sgermanserrano, 2019-01-07:
Adding #1594

isamu-takagi, 2019-01-10:
Adding #1637

hakuturu583, 2019-01-10:
Adding #1860

hakuturu583, 2019-01-17:
Adding #1893

kfunaoka, 2019-01-17:
@kitsukawa requested to add #1905

YamatoAndo, 2019-01-23:
Adding #1788

YamatoAndo, 2019-01-23:
Adding #1895 #1912

kitsukawa, 2019-01-23:
Adding #1792

mitsudome-r, 2019-01-25:
add #1877

filiperinaldi, 2019-01-25:
What is the time frame for contributing towards  v1.11?

yukkysaito, 2019-01-28:
Adding https://github.com/CPFL/Autoware/issues/1907

kfunaoka, 2019-01-28:
Since release/1.10.0 is branched from f77b78492 commited on Jan. 8th,
I propose the next release schedule is shifted 2 weeks later than before.

| Version | Pull Request | Merged | Release |
|-----------|------------|------------|------------|
| v1.11 | Feb. 28th, 2019 | Mar. 7th, 2019 | Mar. 14th, 2019 |

yn-mrse, 2019-01-30:
Adding #1470 #1475 #1940

mitsudome-r, 2019-02-04:
Adding #1954

hakuturu583, 2019-02-04:
Adding #1943

amc-nu, 2019-02-07:
Adding #1966

hakuturu583, 2019-02-12:
#1970

filiperinaldi, 2019-02-15:
Can #1946 be included please.

hakuturu583, 2019-02-19:
Adding #2003 ,#2005

s-azumi, 2019-02-22:
Adding #2001

hakuturu583, 2019-02-22:
Adding #2012

sgermanserrano, 2019-02-25:
The following is an overview of the current state of PRs for new features that are meant to be included in v1.11. Taking into account the amount of work that is still needed for some of them and the limited time for v1.11, can the current developers make a decision as to whether the PRs will be ready in time for appropriate review?

| PR / Issue | Developer | Approver | Notes |
| :----- | :-----: | :-----: | :----- | 
| Feature/rebuild decision maker #1609 #2001 | @kfunaoka @s-azumi  |  N/A |  Tests under development for the different packages of this PR in #1757 #1744 #1958 #2009 | 
| Initial release of Shape Estimation #1815 | @yukkysaito @amc-nu |  @amc-nu |  Unit tests were added as part of the PR | 
| Initial release of Baidu's LiDAR based CNN segmentation #1800 | @amc-nu |  @cirpue49 |  **NO tests** were included for this feature prior to merging |  
| Initial release of Object Filter #1645 | @amc-nu |  N/A |  Closed and moved to #1815 | 
| Initial release of FireSSD image based detector for CPU #1794 | @amc-nu |  N/A |  No associated PR for this issue |  
| Improve Hybrid A* planner #1594  | @aohsato |  N/A | Moved to 1.11 from 1.10. Some tests added in #1753, still missing tests for freespace_planner and astar_avoid | 
| Initial release of Autoware Launcher #1637 #1950 | @isamu-takagi |  @hakuturu583 |  Some tests added as part of [commit](https://github.com/CPFL/Autoware/pull/1950/commits/c4f1b76e74570a969f11eb5de3c3e04131a214a1) | 
| Set negative esimate_twist.twist.linear.x if going backwards #1788 then #1798 | @YamatoAndo |  @kitsukawa |  Merged with *no test* added for new method |  
| Can not reach the end of waypoint #1470 and #1477  | @yn-mrse |  N/A |  No tests added yet | 
| Simplified configuration of waypoint functions #1475 and #1951 | @yn-mrse |  @hakuturu583 |  **NO tests** were included for this feature prior to merging | 
| Enable backward driving #1940 | @yn-mrse |  N/A |  No associated PR for this issue  | 
| [Feature] costmap_generator #1772 and #1774 | @cirpue49 |  N/A |  Tests added, awaiting review  | 
| [Feature] Robust Adavitive Uncented Kalman Filter + IMM-UKF-PDA #1461 and #1685 | @cirpue49 |  N/A |  Tests added, awaiting review  | 
| Feature/vehicle description #1848 | @hakuturu583 |  @yukkysaito |  **NO tests** were included for this feature prior to merging  | 
| Feature/lgsvl sim configurator #1893 | @hakuturu583  |  @yn-mrse |  **NO tests** were included for this feature prior to merging  | 
| [Features] Mobileeye Driver #1905 | @kitsukawa |  N/A |  No associated PR for this issue  | 
| Feature/RosbagController #1791 | @ChenxiTU @kitsukawa |  @isamu-takagi |  **NO tests** were included for this feature prior to merging  | 
| [Feature] feature/nmea_to_mgrs #1877 and #1878 | @mitsudome-r |  N/A |  Closed  | 
| [Feature] Gazebo simulator #1907 and #1930  | @yukkysaito | @aohsato|  **NO tests** were included for this feature prior to merging   | 
| [Feature] Point Pillar for 3D bounding box detection from LIDAR data #1942 | @cirpue49 |  N/A |   No associated PR for this issue   | 
| [Feature] Load other map formats #1954 | @mitsudome-r |  N/A |  No associated PR for this issue  | 
| Feature/autoware health checker #1943, related to #1970, #2003, #2004, #2005, #2006 | @hakuturu583 |  @yn-mrse |  Some tests added. Bug reported on #2012 |

YamatoAndo, 2019-02-28:
Adding  #1384

gbiggs, 2019-03-01:
Why?

hakuturu583, 2019-03-01:
Sorry for my late...
This is fix commit for ./run script.
#2050

isamu-takagi, 2019-03-01:
Adding #1950 to support the latest develop branch and some update.

ChenxiTU, 2019-03-08:
Adding #2046 to expand functions of rviz plugin - rosbag controller.

sgermanserrano, 2019-03-08:
Hi all,
No more new PRs for v1.11 please. There is limited time for appropriate reviews.
I'd also ask contributors for the current PRs that are listed here to assess whether their PRs are truly merge-able or need extra work and should be pushed for next releases.
Thanks!

kfunaoka, 2019-03-11:
Hi all and @aohsato @cirpue49 @sgermanserrano @gbiggs Sorry for the delay of releasing v1.11. I'd like to finalize pull request that can be merged to v1.11. Since I guess checking all pull request in this issue takes a long time, I'd like to discuss some of them can be merged or not in this issue, depending on their completion rate. I heard @aohsato and @cirpue49 are trying to creating tests and documentation of Hybrid A* planner #1594 and costmap_generator #1774 for v1.11.

@aohsato @cirpue49 Would you approximate when these pull request can be ready to merge?

cirpue49, 2019-03-11:
Thank you for your consideration.
I already wrote tests for #1774. Waiting for @sgermanserrano comment.

kfunaoka, 2019-03-12:
I heard some packages that depend on executing shell scripts such as #2109 do not work fine because its execute permission is removed at copying files to install directory.

@hakuturu583 Does LGSVL simulator require the fix similar to #2109?

aohsato, 2019-03-14:
@kfunaoka @sgermanserrano 
Thanks for your discussion about releasing. In #1594, I merged @sgermanserrano 's test code by #1753 and added some fix to solve some test problems, so the library level unit test was completed.
Next, I tried to some test to nodes (10f93e3) but it's difficult to write without our standard code because there are mainly depended on pub/sub/tf system and complex perception results.
If possible, I want to make continuous changes based on the test code policy of v.1.12 or later.

kfunaoka, 2019-03-14:
@aohsato We're now focusing on unit tests within not ROS level but library level. We can go forward to ROS level test and integration test after v1.11 or later. Once the CI in #1594 is passed, would you declare that #1594 was tested on real car, and them merge into develop?

aohsato, 2019-03-14:
@kfunaoka I already tested on our real car, so I additionally comment to PR. CI seems to be passed, so I'll merge. Thanks much!

kfunaoka, 2019-03-14:
@aohsato Fantastic! Autoware is no longer a waypoint follower.

kfunaoka, 2019-03-14:
Hi all,

It seems we can finalize pull request that can be merged to v1.11.

I've created release/1.11.0 branch.
https://github.com/CPFL/Autoware/tree/release/1.11.0

If anyone find something for v1.11.0, please create pull requests to release/1.11.0.

isamu-takagi, 2019-03-14:
@kfunaoka @hakuturu583 @yukkysaito 
Autoware Launcher in develop branch does not work with colcon.
Please review and merge the latest branch #2056.

isamu-takagi, 2019-03-15:
@kfunaoka 
Sorry for very late. Autoware Launcher is merged.

sgermanserrano, 2019-03-15:
@kfunaoka Fresh images have been added to [https://hub.docker.com/r/autoware/autoware/](https://hub.docker.com/r/autoware/autoware/)

kfunaoka, 2019-03-18:
For release note:

- [Fix] rosbag split size in runtime manager (#1817)
- [Fix] negative max_v values for negative omega (#1880)
- Merge pull request #1815 from CPFL/feature/shape_estimation
- Feature/lgsvl sim configurator #1893
- [Feature] negative estimate twist (#1798)
- [Fix] geo_pos_conv class member variable not initialized #1913
- Feature/RosbagController (#1791)
- Changed Autoware/ros/run from being dependent on gksu to using sudo (#1918)
- Feature/autoware health checker (#1943)
- [Feature] waypoint replanner and extractor (#1951)
- [Feature] enable display errors (#1970)
- Fix/remove velodyne subtree (#1966)
- [fix] Install commands for all the packages (#1861)
- [feature] Baidu's CNN based LiDAR segmentation (#1800)
- Feature/spinnaker (#1984)
- Fix/colcon build (#2000)
- [Feature] Autoware Launcher (#1950)
- Feature/autoware health analyzer (#2004)
- [Feature] Improve AS vehicle interface (#1945)
- Merge pull request #1930 from CPFL/feature/gazebo_simulator
- Merge pull request #2020 from CPFL/fix/spinnaker
- Remove catkin_make_release* scripts (#2042)
- Update wiki demo (#2091)
- [Feature]PointPillars (#2029)
- Feature/rosbag_controller_split (#2046)
- [Feature] Rebuild decision maker (#1609)
- [Feature] costmap generator (#1774)
- Merge pull request #2109 from CPFL/fix/gazebo_with_colcon
- Docker cleanup (#1946)
- Removing CUDA dependencies for Darknet Yolov3 (#1784)
- [Feature] Improve Hybrid A* planner (#1594)
- [fix] not predicting objects at x<=0 or y<=0 (#2122)

kfunaoka, 2019-03-21:
Release Note

# Announcement
- Obsolete packages and nodes were removed as announced at [Ver.1.10.0](https://github.com/CPFL/Autoware/releases/tag/1.10.0) (#1812)
- Please use colcon for source build. Catkin is no longer supported (#2042)
The following commands are required once, when you build from source
```
sudo apt-get install -y python3-pip python3-setuptools
sudo pip3 install -U setuptools
sudo apt-get install -y python3-colcon-common-extensions
```

# New Features
- PointPillars: Fast Encoders for Object Detection from Point Clouds (#2029)
https://drive.google.com/file/d/1l6Grsg0JbBX5s7-JQ6eXQY169PCewJnW/view
- Baidu's CNN based LiDAR segmentation (#1800)
https://www.youtube.com/watch?v=fw7522gDBAc
- Shape/Pose Estimation based on PointCloud (#1815)
- Autoware Health Checker (#1943, #1970, #2004)
- ROS driver for pointgrey cameras using Spinnaker SDK (#1984)
- RosbagController (#1791, #2046)

# Experimental Features
- Autoware Launcher provides simple easy-to-use user interface  (#1950)
- Decision maker is rebuilt based on real-world's requirements (#1609)

# Improvements
- Improved Hybrid A* (#1594) and Cost Map using prediction (#1774)
- Removed CUDA dependencies for Darknet (#1784)
- Improved AS vehicle interface (#1945)
- Gazebo Simulator can be launched from Runtime Manager (#1930)
- LVSVL Simulator can be launched from remote machine (#1893)
- Docker cleanup (#1946)
- Divided waypoint_loader into waypoint_loader, waypoin_replanner, and waypoint_extractor (#1951)

# Bug Fixes
- Fixed rosbag split size in runtime manager (#1817)
- Fixed max velocity on right turns at twist_filter (#1880)
- Fixed first topic of gnss_pose that kills ndt_matching (#1913)
- Fixed no prediction about objects at x<=0 or y<=0 (#2122)
- Inserted install commands for all the packages (#1861)

closed 2019-03-23 by kfunaoka