autowarefoundation/autoware #1823: How does yolo replace dpm

tigersujie, 2018-12-28:
I use object-detection by dpm，but the effect is not good。And，the yolo-v3 worked very well。So I want to use object-detection by yolo。But it seems that yolo does not send topics（/obj_xxx/image_obj） to node(/obj_xxx/ranging).So what should I do?thanks~~

yukitsuji, 2018-12-28:
You should check the README.md in the package of vision darknet detect
https://github.com/CPFL/Autoware/tree/master/ros/src/computing/perception/detection/vision_detector/packages/vision_darknet_detect

tigersujie, 2018-12-28:
> You should check the README.md in the package of vision darknet detect
> https://github.com/CPFL/Autoware/tree/master/ros/src/computing/perception/detection/vision_detector/packages/vision_darknet_detect

Thank you very much for your reply!The Video(https://www.youtube.com/watch?v=pO4vM4ehI98) shows that yolo can publish /obj_xxx/image_obj, but in my practice it is found that yolo publishes /detection/vision_objects, the topic will be subscribed to /obj_xxx/ranging, and used to track?Thank you again for your reply~~

amc-nu, 2019-01-04:
DPM is now considered deprecated, and will be removed.
Just as @yukitsuji mentioned
The output is shown here:
https://github.com/CPFL/Autoware/tree/master/ros/src/computing/perception/detection/vision_detector/packages/vision_darknet_detect#published-topics

Fusion is also changing. Please follow:
https://github.com/CPFL/Autoware/issues/1501
for updates.

closed 2019-01-04 by amc-nu