autowarefoundation/autoware #1849: Unable to build docker image for v1.10.0 RC1

kfunaoka, 2019-01-09:
# Bug
## Expected Behavior

Built successfully.

## Actual Behavior

The link failure at vision_dpm_ttic_detect:

```
$ ./build.sh kinetic

...

Starting >>> vision_dpm_ttic_detect
Finished <<< ymc [9.83s]
Starting >>> vision_ssd_detect
--- stderr: multi_lidar_calibrator
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** visualization features related to openni2 will be disabled
---
Finished <<< multi_lidar_calibrator [1min 8s]
Starting >>> decision_maker
Finished <<< vision_beyond_track [27.0s]
Starting >>> fake_autoware_nodes
--- stderr: pixel_cloud_fusion
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** visualization features related to openni2 will be disabled
---
Finished <<< pixel_cloud_fusion [57.6s]
Starting >>> lidar_localizer
--- stderr: vision_ssd_detect
'SSD/Caffe' is not installed. 'ssd_unc' will not be built.
---
Finished <<< vision_ssd_detect [3.70s]
Starting >>> object_map
--- stderr: vision_dpm_ttic_detect
In file included from /home/autoware/Autoware/ros/src/computing/perception/detection/vision_detector/packages/vision_dpm_ttic_detect/nodes/vision_dpm_ttic_detect/vision_dpm_ttic_detect.cpp:28:0:
/home/autoware/Autoware/ros/install/libdpm_ttic/include/libdpm_ttic/dpm_ttic.hpp:24:19: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11
  DPMTTICParam() = default;
                   ^
/usr/bin/ld: warning: libcuda.so.1, needed by /home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so, not found (try using -rpath or -rpath-link)
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuMemFree_v2'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuEventCreate'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuLaunchKernel'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuStreamSynchronize'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuArray3DCreate_v2'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuTexRefSetFilterMode'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuEventElapsedTime'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuCtxSynchronize'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuDeviceGetAttribute'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuMemcpyDtoH_v2'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuTexRefSetAddressMode'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuCtxSetCurrent'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuCtxCreate_v2'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuEventSynchronize'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuModuleUnload'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuMemcpy3D_v2'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuDeviceGet'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuMemsetD32_v2'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuTexRefSetAddress_v2'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuMemFreeHost'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuStreamCreate'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuTexRefSetFormat'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuDeviceGetCount'
/home/autoware/Autoware/ros/install/libdpm_ttic/lib/liblibdpm_ttic.so: undefined reference to `cuModuleGetTexRef'
[... 280 lines cut ...]
[297.728s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_pipe2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:238> wait_for=<Future pending cb=[Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[297.728s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_pipe2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:238> wait_for=<Future pending cb=[Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[297.728s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_pipe2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:238> wait_for=<Future pending cb=[Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[297.728s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_pipe2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:238> wait_for=<Future pending cb=[Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[297.728s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_pipe2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:238> wait_for=<Future pending cb=[Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
[297.728s] ERROR:asyncio:Task was destroyed but it is pending!
task: <Task pending coro=<_pipe2callback() done, defined at /usr/lib/python3/dist-packages/colcon_core/subprocess.py:238> wait_for=<Future pending cb=[Task._wakeup()]> cb=[_wait.<locals>._on_completion() at /usr/lib/python3.5/asyncio/tasks.py:414]>
The command '/bin/sh -c /bin/bash -c 'source /opt/ros/kinetic/setup.bash; cd /home/$USERNAME/Autoware/ros/src; git submodule update --init --recursive; cd ../; ./colcon_release'' returned a non-zero code: 2
```

## Steps to Reproduce the Problem

  1. cd Autoware/docker/generic
  1. ./build.sh kinetic

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: release/1.10.0

esteve, 2019-01-09:
@kfunaoka I believe it's because of the stricter link flags introduced in `autoware_build_flags`

kfunaoka, 2019-01-09:
Thanks! I got it!