autowarefoundation/autoware #1853: Add missing install target to all packages that produce libraries or executables

esteve, 2019-01-09:
In order for colcon to create a proper workspace that can be used, we need to install the produced binaries via the install CMake command.

These are the packages that declare either an (`add_executable`) or a (`add_library`), but don't have an install command associated:

- [ ] src/actuation/vehicles/packages/ymc
- [ ] src/computing/perception/localization/packages/orb_localizer
- [ ] src/computing/perception/localization/packages/orb_localizer/Thirdparty/DBoW2
- [ ] src/computing/perception/localization/packages/orb_localizer/Thirdparty/Pangolin
- [ ] src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o
- [ ] src/computing/planning/mission/packages/freespace_planner
- [ ] src/computing/planning/mission/packages/op_global_planner
- [ ] src/computing/planning/mission/packages/way_planner
- [ ] src/computing/planning/motion/packages/astar_planner
- [ ] src/computing/planning/motion/packages/dp_planner
- [ ] src/computing/planning/motion/packages/ff_waypoint_follower
- [ ] src/computing/planning/motion/packages/lattice_planner
- [ ] src/computing/planning/motion/packages/op_local_planner
- [ ] src/computing/planning/motion/packages/op_simulation_package
- [ ] src/computing/planning/motion/packages/op_utilities
- [ ] src/computing/planning/state/state_machine
- [ ] src/data/packages/obj_db
- [ ] src/driveworks/packages/autoware_driveworks_interface
- [ ] src/sensing/drivers/camera/packages/vectacam
- [ ] src/sensing/drivers/gnss/packages/garmin
- [ ] src/sensing/drivers/gnss/packages/nmea_navsat
- [ ] src/sensing/drivers/imu/packages/memsic
- [ ] src/sensing/drivers/imu/packages/microstrain
- [ ] src/sensing/drivers/lidar/packages/hokuyo
- [ ] src/sensing/drivers/lidar/packages/robosense/rslidar_driver/src
- [ ] src/sensing/drivers/lidar/packages/robosense/rslidar_pointcloud/src
- [ ] src/sensing/drivers/lidar/packages/sick/lms5xx
- [ ] src/sensing/filters/packages/points_downsampler
- [ ] src/sensing/fusion/packages/multi_lidar_calibrator
- [ ] src/sensing/fusion/packages/scan2image
- [ ] src/sensing/polygon/packages/points2polygon
- [ ] src/socket/packages/oculus_socket
- [ ] src/socket/packages/tablet_socket
- [ ] src/socket/packages/udon_socket
- [ ] src/socket/packages/vehicle_socket
- [ ] src/system/sync
- [ ] src/util/packages/data_preprocessor
- [ ] src/util/packages/fake_drivers
- [ ] src/util/packages/kitti_pkg/kitti_player
- [ ] src/util/packages/sample_data

esteve, 2019-01-09:
@amc-nu @kfunaoka @sgermanserrano there's a bunch of packages that need their `CMakeLists.txt` updated before we can remove catkin entirely (see list above). I'll submit PRs for them, but feel free to take some of them too :-)

There's a chance that other packages that have an install command, don't install all their binaries, but we'll have to analyze them one by one.

esteve, 2019-01-09:
@amc-nu @kfunaoka @sgermanserrano there's a bunch of packages that need their CMakeLists.txt updated before we can remove catkin entirely (see list above). I'll submit PRs for them, but feel free to take some of them too :-)

There's a chance that other packages that have an install command, but don't install all their binaries, so we'll have to analyze them one by one.

sgermanserrano, 2019-01-09:
This ties in together with #1812 as some of the listed packages are classified as obsolete in the [spreadsheet](https://docs.google.com/spreadsheets/d/1wcMiQJFr3SwHAS9H21UbcmnmMMDvq04Jpa7uGml0MdI/edit)

amc-nu, 2019-01-09:
It’s worth noting that not only the install command associated to the library or executable is necessary, but also the launch files, configuration files, or any other external files.
We need your kind cooperation 
@aohsato @kitsukawa @messi49 @h-ohta @alexandrx @drwnz @jacoblambert @shinpei0208 @yukkysaito @yk-fujii @hakuturu583 @cirpue49 @daifuku07 @isamu-takagi 
Only each one of you know which files are required in your nodes.

esteve, 2019-01-09:
@amc-nu good point, I've only listed those that had an `add_executable` or `add_library` command, but not the launch files.

I've created https://github.com/CPFL/Autoware/tree/fix/add_install_command so we all can push the changes there.

amc-nu, 2019-01-10:
@esteve @sgermanserrano I fixed all the packages I know about. However, before pushing and updating this issue. I would like to ask for your opinion regarding the installation target path.
Either `CATKIN_PACKAGE_BIN_DESTINATION` or `CATKIN_GLOBAL_BIN_DESTINATION`.
I'm inclined to the `CATKIN_PACKAGE_BIN_DESTINATION`, but I would like to hear your points of view based on your experience.

amc-nu, 2019-01-10:
## Fixed the following launch files and resources

- [X] run (launch UI script)
- [X] src/computing/perception/detection/fusion_tools/packages/pixel_cloud_fusion/CMakeLists.txt
- [X] src/computing/perception/detection/fusion_tools/packages/range_vision_fusion/CMakeLists.txt
- [X] src/computing/perception/detection/lidar_detector/packages/lidar_euclidean_cluster_detect/CMakeLists.txt
- [X] src/computing/perception/detection/lidar_detector/packages/lidar_fake_perception/CMakeLists.txt
- [X] src/computing/perception/detection/lidar_detector/packages/lidar_naive_l_shape_detect/CMakeLists.txt
- [X] src/computing/perception/detection/lidar_tracker/packages/lidar_imm_ukf_pda_track/CMakeLists.txt
- [X] src/computing/perception/detection/lidar_tracker/packages/lidar_kf_contour_track/CMakeLists.txt
- [X] src/computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeLists.txt
- [X] src/computing/perception/detection/vision_detector/packages/vision_dpm_ttic_detect/CMakeLists.txt
- [X] src/computing/perception/detection/vision_detector/packages/vision_segment_enet_detect/CMakeLists.txt
- [X] src/computing/perception/detection/vision_detector/packages/vision_ssd_detect/CMakeLists.txt
- [X] src/computing/perception/detection/vision_tracker/packages/vision_beyond_track/CMakeLists.txt
- [X] src/computing/perception/detection/visualizers/packages/detected_objects_visualizer/CMakeLists.txt
- [X] src/computing/perception/detection/visualizers/packages/integrated_viewer/CMakeLists.txt
- [X] src/computing/perception/detection/visualizers/packages/viewers/CMakeLists.txt
- [X] src/computing/perception/localization/lib/ndt_tku/CMakeLists.txt
- [X] src/computing/perception/prediction/motion_predictor/packages/naive_motion_predict/CMakeLists.txt
- [X] src/computing/perception/semantics/packages/object_map/CMakeLists.txt
- [X] src/computing/perception/semantics/packages/road_occupancy_processor/CMakeLists.txt
- [X] src/computing/perception/semantics/packages/road_occupancy_processor/include/road_occupancy_processor.h
- [X] src/computing/perception/semantics/packages/road_occupancy_processor/src/road_occupancy_processor.cpp
- [X] src/computing/perception/semantics/packages/road_occupancy_processor/src/road_occupancy_processor_node.cpp
- [X] src/computing/planning/decision/packages/decision_maker/CMakeLists.txt
- [X] src/computing/planning/mission/packages/lane_planner/CMakeLists.txt
- [X] src/computing/planning/motion/packages/waypoint_maker/CMakeLists.txt
- [X] src/data/packages/pos_db/CMakeLists.txt
- [X] src/sensing/drivers/camera/packages/baumer/CMakeLists.txt
- [X] src/sensing/drivers/camera/packages/pointgrey/CMakeLists.txt
- [X] src/sensing/drivers/camera/packages/vectacam/CMakeLists.txt
- [X] src/sensing/drivers/camera/packages/vectacam/scripts/vectacam.launch
- [X] src/sensing/filters/packages/image_processor/CMakeLists.txt
- [X] src/sensing/filters/packages/points_downsampler/CMakeLists.txt
- [X] src/sensing/filters/packages/points_preprocessor/CMakeLists.txt
- [X] src/sensing/fusion/packages/multi_lidar_calibrator/CMakeLists.txt
- [X] src/util/packages/autoware_rviz_plugins/CMakeLists.txt
- [X] src/util/packages/kitti_pkg/kitti_box_publisher/CMakeLists.txt
- [X] src/util/packages/kitti_pkg/kitti_launch/CMakeLists.txt
- [X] src/util/packages/kitti_pkg/kitti_player/CMakeLists.txt
- [X] src/util/packages/map_tools/CMakeLists.txt
- [X] src/util/packages/runtime_manager/CMakeLists.txt
- [X] src/util/packages/runtime_manager/package.xml
- [X] src/util/packages/runtime_manager/scripts/qs.yaml
- [X] src/util/packages/runtime_manager/scripts/setup_vehicle_info.launch

sgermanserrano, 2019-01-10:
@amc-nu as per [documentation](http://docs.ros.org/api/catkin/html/user_guide/variables.html) I would use `CATKIN_PACKAGE_BIN_DESTINATION`. 
@esteve thoughts?

esteve, 2019-01-10:
@sgermanserrano yes `CATKIN_PACKAGE_BIN_DESTINATION` is the correct one. Once we migrate to ament, `CATKIN_PACKAGE_BIN_DESTINATION` gets replace by just `bin`, I'll write a guide on how to port our packages to ament.

amc-nu, 2019-02-20:
fixed on #1861

closed 2019-02-20 by amc-nu