autowarefoundation/autoware #1862: Move Docker files to a separate repository

esteve, 2019-01-10:
We're currently storing all the Dockerfiles in the same repository as the codebase, but they could be spun off so we can automate the process of building the images.

For example, we could let Dockerhub or Travis build them and push them to Dockerhub. We're manually building them, which is more error-prone.

What do you think @amc-nu @kfunaoka @sgermanserrano ?

sgermanserrano, 2019-01-10:
@esteve I think that can also be tied to the CI loop with Gitlab, we can set it up to build the image on release versions

kfunaoka, 2019-01-11:
@esteve @sgermanserrano Super nice! We will be able to focus on development instead of building Autoware :)

esteve, 2019-01-11:
Excellent! @sgermanserrano yeah the idea is that the CI will build the images for us and push them to Dockerhub nightly or every time there's a commit.

I don't have permits to create a new repo under the CPFL (or should we start using the Autoware organization?). I think `autoware_docker` would just be fine. Thanks.

sgermanserrano, 2019-01-11:
@esteve do we really need a new repo?

esteve, 2019-01-11:
@sgermanserrano perhaps not, but currently there's so much stuff in the Autoware repo that sometimes it's hard to navigate. We could possibly continue using this repo and then move to a new one if we need it.

esteve, 2019-01-11:
Now I remember why I thought we needed a new repository! Dockerhub can build the image for you, but it needs a certain directory structure. If we have Gitlab do it for us, we might not need a new repo, though.

amc-nu, 2019-01-11:
Do you know what do we need to get featured in
https://github.com/docker-library
?

esteve, 2019-01-11:
@amc-nu I think we just have to submit a pull request https://github.com/docker-library/official-images/pulls

bdholt1, 2019-01-23:
Does this mean that we should be removing dockerfiles from CPFL/Autoware? Or are we going to have the case where we have dockerfiles in 2 places?

closed 2020-05-29 by gbiggs