autowarefoundation/autoware #1863: Moving the origin of map frame

4n00815, 2019-01-11:
Good Day everyone,

Is there a way to move origin of map frame that was created using ndt_mapping node, with the respect to map? At the moment, origin is at the position where ROSBAG recording started.

Thank you,

Strahinja

mitsudome-r, 2019-01-11:
You can move the origin using pcl_transform_point_cloud in pcl-tools.
$ sudo apt install pcl-tools
$ pcl_transform_point_cloud input.pcd output.pcd -trans dx,dy,dz -quat w,x,y,z

If you want to know details about options do  
$ pcl_transform_point_cloud -h

4n00815, 2019-01-11:
Thanks you for your reply. I tried to use package that you suggested, but I keep getting this message:

pcl_transform_point_cloud 6-17.pcd 6-17new.pcd -trans 11672.0970008, 10145.1109782, 99.931  -quat 0.997921317917, 0.0, 0.0, -0.0644441094796

> [parse_3x_arguments] Number of values for -trans (2) different than 3!
> Wrong number of values given (2): The quaternion specified with -quat must contain 4 elements (w,x,y,z).

I think I'm passing the right number of arguments?

mitsudome-r, 2019-01-11:
You might need to  without space in between.
pcl_transform_point_cloud 6-17.pcd 6-17new.pcd -trans 11672.0970008,10145.1109782,99.931 -quat 0.997921317917,0.0,0.0,-0.0644441094796

HHF111, 2020-03-25:
@mitsudome-r @4n00815  Thanks for sharing. Could you please introduce a bit how you calculate -quat w, x, y,z  the four parameters here to align PCD map to UTM coordinate? Thanks!

Pallav1299, 2020-04-06:
> @mitsudome-r @4n00815 Thanks for sharing. Could you please introduce a bit how you calculate -quat w, x, y,z the four parameters here to align PCD map to UTM coordinate? Thanks!

These represent the initial orientation at the vehicle start position. The pointcloud does not include this orientation data until and unless you give it as an input before mapping, considering you are using NDT mapping.
So the main motive is to transform this pointcloud to that initial position and orientation.

This "quat w, x, y,z" value i.e. the orientation will generally come from an IMU. The first orientation value that you receive from it.