autowarefoundation/autoware #1865: [Bug] Unable to launch Runtime Manager with colcon

kfunaoka, 2019-01-11:
# Bug
## Expected Behavior

Runtime Manager appears.

## Actual Behavior

Only the window of roscore appears.

## Steps to Reproduce the Problem

  1. `cd Autoware/ros`
  1. `./colcon_release`
  1. `./run`

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic 
  - Autoware branch: master (v1.10.0) and develop (after c4a3ecc174f066d84d2f45991074d978d0761c42)

amc-nu, 2019-01-11:
@kfunaoka this is expected since the required files of almost all the packages are not being installed.
Please check #1853 and #1861

kfunaoka, 2019-01-11:
@amc-nu Yes, because it is expected, we need this issue for heads-up :)
This issue is referenced from the release note of v1.10.0.
I've added an explanation in the headline.

amc-nu, 2019-02-20:
fixed #1861

closed 2019-02-20 by amc-nu