autowarefoundation/autoware #1881: [Design] frame naming conventions

hakuturu583, 2019-01-12:
Now, autoware has no naming conventions in frame.
So, I want to define frame naming conventions clearly.
Please give us opinions!!

# Design frame naming conventions
[image]
[image]
[image]

msmcconnell, 2019-01-13:
Great to see some frame standardization going on!
I think when possible we should try to conform to ROS [REP 105](http://www.ros.org/reps/rep-0105.html). This will allow other ros users to easily understand our frames (map, base_link, etc.)

I don't think this changes what you currently have proposed with base_link as the vehicle back axle, but I want to point out that ROS has this standard to start from.

mitsudome-r, 2019-01-15:
@msmcconnell Thank you for pointing out.
From my understanding, major difference between Autoware frame and REP 105 is that Autoware does not use "odom" and "earth" frame. This is because these frames are not mandatory in current Autoware implementation. 
I think we shouldn't change naming for commonly used frames in Autoware(i.e. "world", "map", "base_link", "gps") at this point unless there is substantial reason, since many users are used to it now. However, it might be nice to have documentation about these frames somewhere as suggested in "Exception" Section in REP105.

msmcconnell, 2019-01-16:
@mitsudome-r
I think calling out the exceptions to REP 105 would be a great addition to the convention.

mitsudome-r, 2019-01-28:
@hakuturu583 
http://www.ros.org/reps/rep-0103.html
According to above, camera optical frame is defined as follows:
>Suffix Frames
>In the case of cameras, there is often a second frame defined with a "_optical" suffix. This uses a slightly different convention:
>z forward
>x right
>y down

Your optical frame in the image has different orientation.

hakuturu583, 2019-02-04:
fixed camera optical frame.

closed 2020-05-29 by gbiggs