autowarefoundation/autoware #1888: [Question] how to visualize L-shape fitting's output?

tigerk0430, 2019-01-15:
## Specifications

  - Ubuntu version: 16.04 LTS
  - ROS version: kinetic
  - Autoware branch: 1.9.1

Hello Autoware's developers :-) 

Thank you for offering such a nice and great Integrated Packages of Autonomous driving.

I have a question about Visualizing L-Shape Fitting. according to [this video](https://www.youtube.com/watch?v=Zd37mE7sXyI) and [this site](https://github.com/CPFL/Autoware/tree/1.9.0/ros/src/computing/perception/detection/lidar_detector/packages/lidar_naive_l_shape_detect) ,

I can visualize the output of L-shape fitting node `

amc-nu, 2019-01-17:
Please have a look at #1815

tigerk0430, 2019-01-17:
@amc-nu 
Thank you for the reply :-)

I had a look #1815 and realized there are a lot of branches in Autoware project
I've cloned `master` branch and used docker to make it simple.

I just a wonder how to visualize `autoware_msgs/DetectedObjectArray` msg on RViz because there is no plugin to visualize it.
[this](https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/detection/object_filter/packages/lidar_shape_estimation/src/node.cpp#L26) code you suggested is also having a publisher with `autoware_msgs/DetectedObjectArray` and I can't visualized it even though I can run it.

Thank you!

amc-nu, 2019-01-22:
To visualize the `DetectedObjectArray` message you'll need to use the `detected_objects_visualizer` inside perception.

https://github.com/CPFL/Autoware/tree/develop/ros/src/computing/perception/detection/visualizers/packages/detected_objects_visualizer

closed 2019-01-22 by amc-nu

tigerk0430, 2019-01-28:
@amc-nu 

sorry for the late reply 
Thank you! I'll try it out :-)