autowarefoundation/autoware #1906: [Bug] Yolo3 does not work in Docker Image

kfunaoka, 2019-01-17:
# Bug
Yolo3 does not work in Docker Image.
Yolo3 without Docker works fine.

## Expected Behavior
Yolo3 works

## Actual Behavior
Yolo3 outputs the following error message

```
CUDA Error: CUDA driver version is insufficient for CUDA runtime version
```

## Steps to Reproduce the Problem

  1. cd Autoware/docker/generic
  1. ./run -t 1.10.0-kinetic-rc1
  1. cd Autoware/ros
  1. ./run
  1. Rosbag play & Lanuch Yolo3

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: release/1.10.0 (#1898)

In autoware/autoware:1.10.0-kinetic-rc1,
```
$ nvidia-smi
Thu Jan 17 07:39:54 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 396.44                 Driver Version: 396.44                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 108...  Off  | 00000000:01:00.0  On |                  N/A |
| 23%   32C    P8    11W / 250W |    311MiB / 11175MiB |      6%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+

$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2017 NVIDIA Corporation
Built on Fri_Sep__1_21:08:03_CDT_2017
Cuda compilation tools, release 9.0, V9.0.176
```

kfunaoka, 2019-01-17:
I'm resolving the problem with @amc-nu.

- We confirmed that 1.10.0-kinetic-rc1 that I created shows the error.
- I confirmed that Euclidean Cluster on GPU shows the same error -> It seems all CUDA processings are broken.
- I confirmed that the same image based on nvidia/cuda:8.0-devel-ubuntu16.04 (instead of 9.0) also shows the error.

sgermanserrano, 2019-01-17:
@kfunaoka PR #1416 might solve the CUDA issue

kfunaoka, 2019-01-18:
I've created autoware/autoware:1.10.0-kinetic-rc2 based on 5c46036.
I confirmed that RC2 works fine.

@sgermanserrano Thanks. It seems almost reverting b2a6eeb at 258076d solves the problem.

@amc-nu What is the root cause?

amc-nu, 2019-01-21:
@kfunaoka 
These two lines were the root of the problem.
```
RUN sudo ln -s /usr/local/cuda/lib64/stubs/libcuda.so /usr/local/cuda/lib64/stubs/libcuda.so.1
ENV LD_LIBRARY_PATH="/usr/local/cuda/lib64/stubs/:$LD_LIBRARY_PATH"
```

closed 2019-01-21 by amc-nu