autowarefoundation/autoware #1909: Unable to build develop branch

yukkysaito, 2019-01-18:
build error happens with an erroneous description of add dependency.
If cmake do not be written build dependencies correctly, it will not go well or go wrong
```
[ 68%] Generating Python from MSG autoware_msgs/Waypoint
[ 68%] Generating EusLisp code from autoware_msgs/SteerCmd.msg
In file included from /home/yukihiro/workspace/Autoware/ros/src/computing/perception/localization/packages/autoware_connector/nodes/can_status_translator/can_status_translator_node.cpp:20:0:
/home/yukihiro/workspace/Autoware/ros/src/computing/perception/localization/packages/autoware_connector/nodes/can_status_translator/can_status_translator_core.h:29:41: fatal error: autoware_msgs/VehicleStatus.h: No such file or directory
compilation terminated.
[ 68%] Generating C++ code from autoware_msgs/ControlCommandStamped.msg
computing/perception/localization/packages/autoware_connector/CMakeFiles/can_status_translator.dir/build.make:62: recipe for target 'computing/perception/localization/packages/autoware_connector/CMakeFiles/can_status_translator.dir/nodes/can_status_translator/can_status_translator_node.cpp.o' failed
make[2]: *** [computing/perception/localization/packages/autoware_connector/CMakeFiles/can_status_translator.dir/nodes/can_status_translator/can_status_translator_node.cpp.o] Error 1
CMakeFiles/Makefile2:23729: recipe for target 'computing/perception/localization/packages/autoware_connector/CMakeFiles/can_status_translator.dir/all' failed
make[1]: *** [computing/perception/localization/packages/autoware_connector/CMakeFiles/can_status_translator.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 68%] Generating EusLisp code from autoware_msgs/NDTStat.msg
[ 68%] Generating Python from MSG autoware_msgs/DTLane
[ 68%] Generating EusLisp code from autoware_msgs/Signals.msg
[ 68%] Generating EusLisp code from autoware_msgs/TrafficLightResult.msg
[ 68%] Generating EusLisp code from autoware_msgs/ControlCommand.msg
[ 68%] Generating Python from MSG autoware_msgs/DetectedObject
In file included from /home/yukihiro/workspace/Autoware/ros/src/computing/perception/localization/packages/autoware_connector/nodes/can_odometry/can_odometry_node.cpp:20:0:
/home/yukihiro/workspace/Autoware/ros/src/computing/perception/localization/packages/autoware_connector/nodes/can_odometry/can_odometry_core.h:29:41: fatal error: autoware_msgs/VehicleStatus.h: No such file or directory
compilation terminated.
[ 68%] Generating C++ code from autoware_msgs/State.msg
computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/build.make:62: recipe for target 'computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/nodes/can_odometry/can_odometry_node.cpp.o' failed
make[2]: *** [computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/nodes/can_odometry/can_odometry_node.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 68%] Generating EusLisp code from autoware_msgs/ObjLabel.msg
[ 68%] Generating C++ code from autoware_msgs/ImageRect.msg
[ 68%] Generating EusLisp code from autoware_msgs/BrakeCmd.msg
[ 68%] Generating EusLisp code from autoware_msgs/ProjectionMatrix.msg
[ 68%] Generating EusLisp code from autoware_msgs/ImageObjRanged.msg
In file included from /home/yukihiro/workspace/Autoware/ros/src/computing/perception/localization/packages/autoware_connector/nodes/can_odometry/can_odometry_core.cpp:17:0:
/home/yukihiro/workspace/Autoware/ros/src/computing/perception/localization/packages/autoware_connector/nodes/can_odometry/can_odometry_core.h:29:41: fatal error: autoware_msgs/VehicleStatus.h: No such file or directory
compilation terminated.
[ 68%] Generating Python from MSG autoware_msgs/DetectedObjectArray
computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/build.make:86: recipe for target 'computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/nodes/can_odometry/can_odometry_core.cpp.o' failed
make[2]: *** [computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/nodes/can_odometry/can_odometry_core.cpp.o] Error 1
CMakeFiles/Makefile2:23852: recipe for target 'computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/all' failed
make[1]: *** [computing/perception/localization/packages/autoware_connector/CMakeFiles/can_odometry.dir/all] Error 2
```

## Steps to Reproduce the Problem
  1. ./catkin_make_release
## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: develop

yukkysaito, 2019-01-18:
https://github.com/CPFL/Autoware/issues/1859

sgermanserrano, 2019-01-18:
@yukkysaito can you try `catkin_make` directly? without using the script

yukkysaito, 2019-01-18:
@amc-nu just fixed bug.
https://github.com/CPFL/Autoware/commit/b21b663dc4e718e29de96f53efffb766d8af5ebe
This commit will merge into develop soon.

closed 2019-01-22 by amc-nu

kfunaoka, 2019-01-22:
Fixed by merging #1901