autowarefoundation/autoware #1910: Lattice planner / Lane planner topic conflict.

lrodrig3, 2019-01-18:
# Bug

I am trying to run the lattice planner in simulation mode, to understand the capabilities of the planner since I am comparing the performance of different planning algorithms in different AD Stacks. 

During my analysis I realized that the lattice_planner package has a simulation mode in the lattice_velocity_set node subscribes to the topic` \base_waypoints` which are used to compute the lattice. However, the`\base_waypoints` topic is published by the lane_select planner. When I enable the lane_select node of the lane_planner package  it subscribes to the `/state` topic which refers to the configuration state from the state_machine old package. The lattice_trajectory_gen node also publishes a topic called` /state` which is used by the /command_converter and refers to the state (attitude/velocity) of the vehicle. 

[image]

## Expected Behavior

lattice_planner working. 

## Actual Behavior
`[ERROR] [1547801090.778947138]: Client [/lane_select] wants topic /state to have datatype/md5sum [std_msgs/String/992ce8a1687cec8c8bd883ec73ca41d1], but our version has [std_msgs/Float64MultiArray/4b7d974086d4060e7db4613a7e6c3ba4]. Dropping connection.
`

## Steps to Reproduce the Problem

  1. Enable lane_select node
  2. Enable lattice_velocity_set node 
  3. Enable Lattice 

## Specifications

  - Ubuntu version: 16.04
  - ROS version: Kinetic
  - Autoware branch: master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

travelbureau, 2019-01-18:
Lattice planner predates the lane_select planner. Unfortunately, I have not had time to maintain compatibility with recent updates to Autoware. Maybe someone from the team such as @hatem-darweesh can update you on current conventions and his more stable work in 'open-planner'. I think the fix would be relatively simple given accurate information about naming conventions....

lrodrig3, 2019-01-21:
I've changed the topic name of `lattice_trajectory_gen` and in `lattice_twist_convert` to `/vehicle_state` and it solved the issue. However I still cannot run the simulation mode. I'm trying to replicate something similar to the case observed in this [video](https://www.youtube.com/watch?v=QOrsC1P8nN0). What I do are the following steps: 

1. Load Vehicle model and vehicle info from ./config folder
2. Add the Moriyama vector information and PCD from the DEMO files. 
3. Load the TF with the TF information in the demo files.
4.  Enable the `vel_pose_connect` in simulation mode. 
5.  Enable `lane_select` and` lane_rule`
6.  Enable the `wf_simulator` with a pose in RVIZ for localization
7. Enable all the nodes in the` lattice_planner` 
8. Enable `twist_filter. `

After performing these steps I can see in a rqt_graph that all the nodes should have information to start publishing the cubic splines but there is nothing out of it. I wonder if there is something missing or what the issue might be.

lrodrig3, 2019-01-23:
I was able to replicate the demo of the video by creating a global path with the OP global planner, then enable the lane_rule, lane_stop and lane_select nodes which consume the path information from the OP planner and then to enable the lattice planner nodes and the twist filter, this works with the wf simulator. Somehow, the `lane_navi` was not generating any information that would produce a valid `base_waypoint`  topic for the lattice planner. Anyway I will keep working on this and posting here what my findings are.

closed 2020-05-29 by gbiggs

marcusvinicius178, 2021-08-31:
Hi There @lrodrig3  which topic did you subscribe in rviz to see the lattice planner ? 
The car moves, I mean it get the lattice planner as global and local...however I cannot see the lattice as you did in your video.
which plugin and topic do you subscribe to get the visualization