autowarefoundation/autoware #1916: Ask questions about ndt_matching

liuqi8827, 2019-01-23:
Tanks for the great work of your group.  I am facing a problem when I use Autoware(**ndt_matching.launch**) for localization.

## Question
## Expected Behavior
localization successfully.

## Actual Behavior
localization failed.

I had builded a map by using "**approximate_ndt_mapping.launch**", the dataset is my own bag that collected by VLP-16.
When I use "ndt_matching.launch" (launch papramter "**use_odom=true**") for localization, it can run a short distance successfully. But it always failed when the car turn left. 
More specifically, the car is always driving **backwards** when it drives to a fixed place in the pre-build map(In fact, the car should drive **forward**).
[image]

There is a  corresponding video link:
https://drive.google.com/open?id=115St9rP9wzsrprda5Fhna841XmElpPP1

## Specifications

  - Ubuntu version:16.04
  - ROS version:Kinetic
  - Autoware branch:master

I will be very appreciate if you can give any answer.

YamatoAndo, 2019-01-23:
@liuqi8827 Hi.
Is /vehicle/odom topic (nav_msgs::Odometry) included in rosbag?
If use_odom is true, ndt_matching subscribes to /vehicle/odom.

liuqi8827, 2019-01-23:
@YamatoAndo Hi,
Thanks for your  prompt reply. **Yes**, I have /vehicle/odom topic which type is "nav_msgs::Odometry".
From the beginning performence of the video, we can learn that my parameter setting is correct. But I don't why it always failed at the **fixed place**. I had tried many times, it failed at the  **fixed place**  In the same way every time.
In my opinion, I **guess** the robustness of ndt-registration **maybe** one of the reason(I just say **maybe** ). Because I can't find out the reason why it ran successfully in the beginning in a short distance then failed.

YamatoAndo, 2019-01-23:
@liuqi8827 Hi.
As you say, the possibility that the robustness of NDT is doing bad is not zero. However, if you can do mapping, you should be able to do matching as well.

Since it is difficult to make judgments simply by exchanging messages in this form, would you please share the point cloud map with rosbag if possible?

By the way, will position estimation fail even if use_odom is set to false?

liuqi8827, 2019-01-23:
@YamatoAndo Hi
Thanks for your reply. I agree with your view.

1.would you please send me your email adress?  I will share the  point cloud map and rosbag to you by email. I **modified** some code for using my own dataset. And I add **map_stitching node** for stitch the submap that generated by **approximate_ndt_mapping.launch**. I will send my code by email for your test. My email is liuqi8827@gmail.com.
2.If **use_odom = false**, the ndt_matching node can't initialize the position of car in map successfully because there is a severe time delay in my computer(maybe the reason is I **modified** some code for using my own dataset). More specifically, I initialize the position of car as following:
[image]
After I click **"pause"** button,  the car should appear **immediately** in Rviz, but there are about **three seconds delay** when the car appeared at the origin of the map in Rviz. In fact, the car should't localized at the origin of the map, it should localized **several meters away** from the origin of the map because of the **time delay**. Actually,  the car should't localized at the origin of the map, but the **/points_raw**  topic data is **three seconds delayed data**, **/points_raw** topic data can't match with the **point cloud** that is nearby the origin of the map. So I must set **use_odom = true** move the car by **/vehicle/odom**. Then the car can localize successfully.

Thanks so much for your help.

YamatoAndo, 2019-01-25:
@liuqi8827 Hi
I understood why you have to set use_odom to true.
I'd like to think about how to improve by looking at your data set and code.

Btw, I mailed you yesterday, have you received it?
My email address is yamato.ando@tier4.jp.

liuqi8827, 2019-01-25:
@YamatoAndo  Hi
Yes, I received your email and I'm sending the data set and code to you by email.

Thanks a lot.

liuqi8827, 2019-01-28:
@YamatoAndo Hi,
With the help of YamatoAndo, the problem solved. 

The reason is the coordinates of  /vehicle/odom topic in my dataset.
In my dataset, /vehicle/odom/twist/twist/linear represent the speed in the x and y directions based on Map coordinates.
However, since Autoware is based on base_link coordinates, linear.x represents vehicle speed and linear.y is always 0.

Tanks a lot.

closed 2019-01-31 by YamatoAndo