autowarefoundation/autoware #1922: Remove license violations

kfunaoka, 2019-01-25:
# Description
Remove license violations

# ToDo
- [ ] https://github.com/CPFL/Autoware/blob/develop/ros/src/computing/perception/localization/lib/ndt_cpu/

kfunaoka, 2019-01-25:
Anyway, copy/paste is a bad idea.
We should push modifications into upstream if necessary.

ChenxiTU, 2019-01-25:
https://github.com/CPFL/Autoware/blob/develop/ros/src/util/packages/autoware_rviz_plugins/rosbag_controller/src/core/recorder.cpp
copied from 
https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rosbag/src/recorder.cpp

https://github.com/CPFL/Autoware/blob/develop/ros/src/util/packages/autoware_rviz_plugins/rosbag_controller/src/core/recorder.h
copied from 
https://github.com/ros/ros_comm/blob/kinetic-devel/tools/rosbag/include/rosbag/recorder.h

Not 100% copy, I add some other functions to support rest codes. But sorry, I don not modify the licenses. Should I append Apache license before original BSD Licenses?

kfunaoka, 2019-01-25:
@ChenxiTU Would you append Apache license before original BSD Licenses? Your contribution is Apache 2.0 👍

esteve, 2019-01-29:
@ChenxiTU I just found https://github.com/CPFL/Autoware/pull/1791#issuecomment-453017046 but there's no explanation why you found it hard to control recording from GUI difficult. In any case, you could have inherited from the original code or encapsulated it, instead of copy/pasting it.

ChenxiTU, 2019-01-29:
@esteve The main change is that I added Recorder::stop() and Recorder::start to control the start and stop of recording. I also modify several lines for subscribing topics and add some parameters to control the status. I agree with you, inheriting from the original code can be a better choice. In the further, I will add play function. Let me rewrite recorder core at that time.

esteve, 2019-01-29:
@ChenxiTU could you describe the changes in a more detailed way? I want to understand why the original code was difficult to integrate for your use case.

ChenxiTU, 2019-01-30:
@esteve 
Add variable `recorder_status_ ` to track the statement of recording
Add variable `recorder_error_;` to record where and what type of error happened but not shut down the node directly. Some corresponding lines also be modified in like `Recorder::checkDuration, Recorder::checkSize(), Recorder::startWriting()` etc. 
Add variable `std::vector<boost::shared_ptr<ros::Subscriber> > subscribers_` to store all `sub` and make sure all `sub` will be shut down in `Recorder::stop()`. Because of this, add one line `subscribers_.push_back(sub);` in `Recorder::subscribe`. Without doing this, while recording second or third bags, plugin may crash.
Add method `Recorder::start(), Recorder::stop()`

esteve, 2019-01-30:
@ChenxiTU thank you so much for the detailed explanation, I have a better idea of the changes now.

closed 2020-05-29 by gbiggs