autowarefoundation/autoware #1923: How to evaluate accuracy of LiDAR localizer ? Is there any way to generate ground truth .

ajay1606, 2019-01-25:
Hello all,

I have successfully tested the localization module in the autoware using our own custom map and bag file. From visualization, it is clear that ndt_matching working well, But I am not sure the accuracy of the matching algorithm.  So how can I evaluate the accuracy of the ndt_matching? How can I consider the ground truth results?

Could anyone please suggest me?

Kind regards,
Ajay

liuqi8827, 2019-01-25:
Hi @ajay1606,

evo is a good tool to evaluate the accuracy of odometry and localization.
https://github.com/MichaelGrupp/evo

[image]

ajay1606, 2019-01-28:
@liuqi8827 Thank you i will go through this .

HongweiSunny, 2019-03-20:
Hello，have you solved your problem?

closed 2019-03-20 by gbiggs