autowarefoundation/autoware #1925: Grasshopper3 camera Error

rorypeck, 2019-01-25:
### roslaunch autoware_pointgrey_drivers grasshopper3.launch
... logging to /home/iae/.ros/log/2cb2fa7e-2079-11e9-886b-10f0052a9351/roslaunch-iae-PC-32690.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://iae-PC:35743/

SUMMARY
========

PARAMETERS
 * /grasshopper/calibrationfile: 
 * /grasshopper/format: raw
 * /grasshopper/fps: 20
 * /grasshopper/mode: 0
 * /grasshopper/timeout: 1000
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    grasshopper (autoware_pointgrey_drivers/grasshopper3_camera)

ROS_MASTER_URI=http://localhost:11311

process[grasshopper-1]: started with pid [32732]
[ INFO] [1548408035.547967883]: fps set to 20
[ INFO] [1548408035.549686902]: mode set to 0
[ INFO] [1548408035.553313516]: format set to raw
[ INFO] [1548408035.554769381]: timeout set to 1000 ms
Number of cameras detected: 1
supported: 1
mode: 0
maxWidth: 1920
maxHeight: 1200
packetSize: 5472
percentage: 82.6087
pixelFormatBitField: 4236115968
Image settings: 
	Mode: 0
	Pixel Format: 0x400000
	Offset X: 0
	Offset Y: 0
	Width: 1920
	Height: 1200
Packet size: 5472 (82.6087%)

*** CAMERA INFORMATION ***
	Serial number       - 18176637
	Camera model        - Grasshopper3 GS3-U3-23S6C
	Camera vendor       - Point Grey Research
	Sendor              - Sony IMX174 (1/1.2" Color CMOS)
	Resolution          - 1920x1200
	Firmware version    - 2.30.3.0
	Firmware build time - Wed Mar 29 00:05:26 2017
[ INFO] [1548408035.624320273]: No calibrationfile param was received
[ INFO] [1548408035.628257701]: Publishing.. image_raw
Error Trace:
Source: IidcCameraInternal.cpp(466) Built: Oct 15 2018 15:25:36 - Error starting isochronous stream.
+-> From: Iso.cpp(2069) Built: Oct 15 2018 15:23:18 - Failed to check setting of power bit.

Capturing by 1 cameras...
Error Trace:
Source: CameraInternal.cpp(129) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
+-> From: CameraInternal.cpp(113) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
Error Trace:
Source: CameraInternal.cpp(129) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
+-> From: CameraInternal.cpp(113) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
Error Trace:
Source: CameraInternal.cpp(129) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
+-> From: CameraInternal.cpp(113) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
Error Trace:
Source: CameraInternal.cpp(129) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
+-> From: CameraInternal.cpp(113) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
Error Trace:
Source: CameraInternal.cpp(129) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
+-> From: CameraInternal.cpp(113) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
Error Trace:
Source: CameraInternal.cpp(129) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.
+-> From: CameraInternal.cpp(113) Built: Oct 15 2018 15:21:17 - Generation count is incorrect.

  1.
  1.
  1.

## Specifications

  - Ubuntu version:
  - ROS version:
  - Autoware branch:

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

amc-nu, 2019-01-25:
Seems like some cable/power related problem

rorypeck, 2019-01-25:
Hi, I short the USB cable, the problems listed above are solved. But I met another problem, the process died at "start_capture(cameras)", how can it be fixed?

amc-nu, 2019-01-25:
Can you share the whole output error/ trace?

rorypeck, 2019-01-25:
well, it has no trace, just said the process has died. And I am very sorry that I have left the lab, and can only send the errors to you by tommorrow.

rorypeck, 2019-01-26:
@amc-nu Hi, Abrahan, the whole output are listed below:

/home/iae/AutoVehicle/autoware/Autoware-1.10.0/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py:2889: wxPyDeprecationWarning: Call to deprecated item. 
  wx.InitAllImageHandlers()
loading param.yaml
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
loading qs.yaml
loading setup.yaml
['rosparam', 'set', 'tf_x', '1.2']
['rosparam', 'set', 'tf_y', '0.0']
['rosparam', 'set', 'tf_z', '2.0']
['rosparam', 'set', 'tf_yaw', '0.0']
['rosparam', 'set', 'tf_pitch', '0.0']
['rosparam', 'set', 'tf_roll', '0.0']
['rosparam', 'set', 'localizer', 'velodyne']
loading map.yaml
loading sensing.yaml
loading computing.yaml
loading interface.yaml
loading data.yaml
loading simulation.yaml
['rosparam', 'set', '/use_sim_time', 'false']
loading status.yaml
loading state.yaml
loading topics.yaml
['roslaunch', 'autoware_pointgrey_drivers', 'grasshopper3.launch', 'mode:=0', 'format:=raw', 'fps:=20', 'timeout:=1000']
pid=8564
sched policy=OTHER prio=0
... logging to /home/iae/.ros/log/a111d2a2-2117-11e9-91e6-10f0052a9351/roslaunch-iae-PC-8564.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://iae-PC:40563/

SUMMARY
========

PARAMETERS
* /grasshopper/calibrationfile:
* /grasshopper/format: raw
* /grasshopper/fps: 20
* /grasshopper/mode: 0
* /grasshopper/timeout: 1000
* /rosdistro: kinetic
* /rosversion: 1.12.14

NODES
/
grasshopper (autoware_pointgrey_drivers/grasshopper3_camera)

ROS_MASTER_URI=http://localhost:11311

[ INFO] [1548472202.968173239]: fps set to 20
[ INFO] [1548472202.969709291]: mode set to 0
[ INFO] [1548472202.971377827]: format set to raw
[ INFO] [1548472202.972963289]: timeout set to 1000 ms
Number of cameras detected: 1
supported: 1
mode: 0
maxWidth: 1920
maxHeight: 1200
packetSize: 5472
percentage: 82.6087
pixelFormatBitField: 4236115968
Image settings:
Mode: 0
Pixel Format: 0x400000
Offset X: 0
Offset Y: 0
Width: 1920
Height: 1200
Packet size: 5472 (82.6087%)

*** CAMERA INFORMATION ***
Serial number       - 18176637
Camera model        - Grasshopper3 GS3-U3-23S6C
Camera vendor       - Point Grey Research
Sendor              - Sony IMX174 (1/1.2" Color CMOS)
Resolution          - 1920x1200
Firmware version    - 2.30.3.0
Firmware build time - Wed Mar 29 00:05:26 2017
[grasshopper-1] process has died [pid 8608, exit code -11, cmd /home/iae/AutoVehicle/autoware/Autoware-1.10.0/ros/devel/lib/autoware_pointgrey_drivers/grasshopper3_camera __name:=grasshopper __log:=/home/iae/.ros/log/a111d2a2-2117-11e9-91e6-10f0052a9351/grasshopper-1.log].
log file: /home/iae/.ros/log/a111d2a2-2117-11e9-91e6-10f0052a9351/grasshopper-1*.log
process[grasshopper-1]: started with pid [8608]
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

rorypeck, 2019-01-26:
I have tested on windows by spinnake viewer and can see the image, I will try it with FlyCapture viewer

rorypeck, 2019-01-26:
@amc-nu Hi, I have test with the FlyCap2 viewer, and it works normally

amc-nu, 2019-01-26:
now please obtain the mode number that the camera supports. By default the launch file is set to 0
You’ll need to change that value to the one supported by your camera. Also you’ll need to confirm that the image format is supported in the selected mode (RGB or RAW).

rorypeck, 2019-01-26:
OK. 
[image]

rorypeck, 2019-01-26:
OK, the mode number supported is 0, 4, 5 and 7, with 0 default. And the image format contains Raw 8 and RGB 8, don't know this should be modified in the source code?

[image]

rorypeck, 2019-01-26:
No, I test Autoware on ubuntu 16.04, and the FlyCapture Viewer can only run on Windows, am I wrong?

amc-nu, 2019-01-26:
The viewer is also included in the Linux installation.

rorypeck, 2019-01-26:
Thanks for your kind and patient help. the problem is solved finally. I find the camera can run normally on mode 4, if changed back to mode 0, errors still occur, I will study further to overcome the problem
[image]

amc-nu, 2019-01-26:
I’m glad to hear it’s working. I’m closing this issue.

closed 2019-01-26 by amc-nu

rorypeck, 2019-03-04:
@amc-nu Finally fix the problem, this is the technical docs:
On Linux systems, image size is restricted to 2MiB or less by default. To
increase this limit so that you can make use of your imaging hardware's full
capabilities, you will need to make a minor change to your system.
More information on these changes can be found at:
<https://www.ptgrey.com/tan/10685#ConfiguringUSBFS>

1. Open the /etc/default/grub file in any text editor. Find and replace:

GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"

with this:

GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=1000"

2. Update grub with these settings:
$ sudo update-grub

3. Reboot and test a USB 3.1 camera.
If this method fails to set the memory limit, run the following command:
$ sudo sh -c 'echo 1000 > /sys/module/usbcore/parameters/usbfs_memory_mb'

To confirm that you have successfully updated the memory limit, run the following command:
$ cat /sys/module/usbcore/parameters/usbfs_memory_mb

amc-nu, 2019-03-04:
I’m glad you solved your problem. Indeed that’s part of the installation procedure of the SDK.
Thanks for sharing.

rorypeck, 2019-03-04:
Yes, I skipped such a step of the SDK installation and met a problem that cannot be understood, so published it here for someone else