autowarefoundation/autoware #1937: [Discussion] Map Architecture

kfunaoka, 2019-01-28:
A map architecture which welcomes multiple map formats/companies/organization/researchers.

[image]

# A.I.
- [ ] Dig down map abstraction layer

kfunaoka, 2019-01-28:
@gbiggs suggested that Internal Preresentation is more suitable for ROS convention.
[image]

hadiTab, 2019-01-29:
@kfunaoka has the internal representation been decided yet?

kfunaoka, 2019-01-29:
Not yet and A.I.

msmcconnell, 2019-01-29:
I think a map abstraction layer is a great idea.  Are you looking to define the API in this issue as well, or just propose the creation of such an API?

gbiggs, 2019-01-30:
> A map abstraction layer is a great idea! Are you looking to define the API in this issue as well, or just propose the creation of such an API?

I think the idea is to define that API here as well.

The information model that the API represents will need to be a superset of all supported map formats. Although the diagram above lists "other format", there is no guarantee that any information model we define will be able to support all information contained in some unknown, possibly as-yet-undefined map format. I propose we define which map formats will be supported now so that we know what information must be representable. We can extend the information model in the future when and if necessary.

I think a good selection that covers maps commonly used and those in use by Tier IV is Aisan, Autoware, Lanelet2 and OpenDRIVE.

TonysCousin, 2019-01-30:
Generally agree, but the API doesn't necessarily need to be a strict superset of all supported formats. It may be worth choosing to ignore a particular feature of a particular format that is seen as low value because it would be seldom used. Eliminating difficult "special interest" cases, if any exist, could allow much faster progress.  Then those cases could be considered for inclusion in a later enhancement.

mitsudome-r, 2019-01-31:
The format proposed in #1701 should contain most of commonly used information in Autoware and should have less "unwanted" information (since that how it was made). Maybe internal representation can start with something similar to the format and extend it to add any valuable features from other formats.

mitsudome-r, 2019-01-31:
Another possible solution that I can think of is to use Lanelet2 as base.
It already has some useful API, and seems to contain most of required information.
My concern is that Lanelet is white line based implementation, while current Autoware is mainly centerline based(waypoint based) so it might require some workload to replace current implementation with Lanelet2-based implementation.

msmcconnell, 2019-01-31:
@mitsudome-r 
LaneLet2 does provide center line calculations per documentation https://github.com/fzi-forschungszentrum-informatik/Lanelet2/blob/master/lanelet2_core/doc/LaneletPrimitives.md
```
lanelets may have an additional centerline to guide the vehicle. This centerline must be within the area formed by the left and right bound and must not touch the boundaries. **If no centerline is given, the library will calculate it for you.**
```
However, the key for an interface like that proposed above is how we represent the lane geometry and connections in ROS messages. I believe that the description should match one of the standards presented in #1701. This ensures that all users have a clearly defined way of interpreting the data. Regardless of which file format or processing component is actually used for map data, a single interface representation should be selected to describe the road in the abstract.

mitsudome-r, 2019-02-01:
@msmcconnell 
Yes, I am aware of the center line calculation.
However, there is no connection information between the center lines of each lanelet. You have to reference back to lanelet connection in order to connect the center line, which I thought is bothersome if we mainly use center line for calculation. Also, points of the centerline doesn't have unique ID, and we have to assign them by our hands.

I have a feeling that we at least need to re-organize or modify Lanelet to some extent in order to use it in Autoware without stress. Once we do that modification, we have to convert them back to Lanelet format to use Lanelet2 functions, which kind of weakens the merit of using Lanelet2 in calculation efficiency. 

> However, the key for an interface like that proposed above is how we represent the lane geometry and connections in ROS messages.

I also think this is very important. I've asked in Lanlet2 if they have such interface, but they said that it will be available later this year. 
https://github.com/fzi-forschungszentrum-informatik/Lanelet2/issues/7

msmcconnell, 2019-02-04:
@mitsudome-r 
Those are valid concerns with regards to LaneLet2 and probably worth sharing in the other issue page as well. 

Why don't we start moving this issue forward by identifying the high level data we need even if some details like (lane id) may not yet be determined. 

For this purpose, I have created the following document. Please leave comments. After we agree upon a data set and access method, we can start outlining ROS messages.
https://docs.google.com/document/d/1bRKtcFLnzv0C-zgzOpggp7TU2MW1y8Q0NgVNhGVSMT4/edit?usp=sharing

closed 2020-05-29 by gbiggs