autowarefoundation/autoware #1938: Autoware Dockerfiles clean up

sgermanserrano, 2019-01-29:
# New Feature
The current level of duplicity within the Dockerfiles can be removed. 

4 cases of usage for the docker images have been identified (without considering cross-compiling):

1. Environment without CUDA support
2. Environment with CUDA support
3. Autoware built without CUDA support
4. Autoware built with CUDA support

The above is translated into 4 Docker images and 3 Dockerfiles:
1. Base image with all non-CUDA related dependencies. (_Dockerfile.cpu.kinetic_dependencies_)
2. Base image with CUDA related dependencies built from 1. (_Dockerfile.cuda.dependencies_)
3. Autoware built on top of 1. (_Dockerfile.autoware_)
4. Autoware built on top of 2. (_Dockerfile.autoware_)

To keep the current state of build, case 4 can be set to be the default one in the _build.sh_ and _run.sh_ scripts.

The split of Dockerfiles should also help at building Autoware docker images automatically.

amc-nu, 2019-01-29:
@Servando

I’ve been working exactly in that 
Please have a look at:

https://github.com/CPFL/autoware_docker_images

The images are implemented in that order with an extra layer. This allows to run within the allowed time in Travis CI and Docker Hub. Please let me hear your opinions, and feel free to change the repository.

sgermanserrano, 2019-03-13:
Solved in #1946

closed 2019-03-13 by sgermanserrano