autowarefoundation/autoware #1940: [Feature] Enable backward control

yn-mrse, 2019-01-30:
# New Feature
Running in the backward direction is essential function for turning back or parking back.
However, it is not yet realized in Autoware now.
Since individual nodes are made on the assumption of running in the forward direction, it is necessary to correct them.

closed 2020-05-29 by gbiggs

kosmastsk, 2020-07-24:
Has any progress been made in this feature?

JWhitleyWork, 2020-07-24:
> Has any progress been made in this feature?

Since this issue was created, yes. However, we have moved to Gitlab and back since then. You can find the Gitlab version of this ticket here: https://gitlab.com/autowarefoundation/autoware.ai/core_planning/-/issues/8. However, this feature was not completed (as you can see from that issue) and there is still more to be done. We would be glad to see PRs for this feature.

d0l1v3, 2022-03-11:
According to issue #8 in GitLab this issue should be resubmitted at core_planning but unfortunately I couldn't find a corresponding issue for that. Has there been any progress? Thanks!

JWhitleyWork, 2022-03-11:
@d0l1v3 Please do not bump zombie issues. Additionally, Autoware.AI is no longer actively developed. This feature will never be added to this fork of Autoware.AI.