autowarefoundation/autoware #1952: How to prepare rosbag file before using is generating 3D map?

Jiaminchen0, 2019-02-02:
Hi,

I generated a rosbag file, but I couldn't use it to generate the map using Autoware. 

The "map" generated by Autoware is like a spiral.

I noticed the demo data has the topic nmea_msgs. So do I need to subscribe to this topic during my record?

Thanks,

YamatoAndo, 2019-02-05:
@Jiaminchen0 Hi, thank you for sharing dataset.

Please set "Mininum Scan Range" to a small value on the [app] tab of [ndt_mapping].
The point where the distance from LiDAR is nearer than "Mininum Scan Range" is not used for mapping.
We usually set LiDAR on top of a car. We set this parameter so that the roof of the car does not remain on the map.

I think you can create a map like the one below. (Mininum Scan Range is 1.0)
[image]

Jiaminchen0, 2019-02-05:
@YamatoAndo Thank you so much for helping me out. 
It worked.

javcursor, 2019-02-08:
Hi, I am in a similar situation. 

At the beginning of my tests I got a map like this
[image]
Yes, it is a karting circuit. This map was obtained with Velodyne VLP16 without any inclination. Recently, it has been necessary tilt 10 degrees so I've had to get a new map.
I am trying to get it with this parameters:
[image]
[image]
But all the pcd that I've obtained are similar to these
[image]
[image]

Data set is here: [https://www.dropbox.com/s/zlilgg6xx6ae1c2/autoware-20190206100413.bag?dl=0](url)

I've made various parameter configuration test without positive results, I need some helps.
Any suggestion?

@YamatoAndo

YamatoAndo, 2019-02-08:
@javcursor Hi.
It is not recommended to tilt sensors when build a map.
These are the reasons:
1. less points are obtained from surrounding objects to be used for matching
2. smaller overlapping scanning range between different time steps 
3. too many points are obtained from ground
I would like to fix this problem, but I still haven't come up with a good way to do it. 

One of the workarounds is to increase minimum scan range so that ground will be ignored.
Following map is created with minimum scan range = 30.

[image]

YamatoAndo, 2019-02-08:
@javcursor 
pcd file: https://www.dropbox.com/s/nq21o8rkte18z4c/autoware-190208.pcd?dl=0

YamatoAndo, 2019-02-09:
@javcursor 
I think that it is better to use "pcl_generic" rather than "pcl_openmp".

javcursor, 2019-02-09:
Thanks you very much for the map and the recommendations @YamatoAndo

closed 2019-02-12 by YamatoAndo