autowarefoundation/autoware #1955: [Feature] MPC for accurate path following

TakaHoribe, 2019-02-04:
Model predictive controller for accurate path tracking.

# New Feature
Implement MPC. The overview of architecture is 
[image]

Although this is implemented in branch,  the following points are still open to discussion.

- Vehicle Model
- QP solver
- Path smoothing filter
- Low pass filter for desired steering and velocity

 branch: https://github.com/CPFL/Autoware/tree/feature/mpc_trajectory_follower

msmcconnell, 2019-02-04:
The mathematical model we use for MPC should be kept separate from the solver if possible. This will allow for different vehicle configurations to be more easily switched out.

TakaHoribe, 2019-02-05:
@msmcconnell yes, that would be best. The MPC solver needs matrix form, so the each model may be generated in the matrix form.
But, what is needed for the matrix depends on the model, for example, current velocity and curvature is needed for the vehicle kinematics model. This should be cared for each model, not generally. so, it's not that easy...
Do you have any good idea or nice example?

TakaHoribe, 2019-02-06:
@msmcconnell 
OK, thanks. I'm going to dig into the way.

QP solver, No, but I will try some to check speed and accuracy. If you have any recommendation, please let me know.
For now, I just solve without constraint problem using inverse matrix, which does not need the QP solver.

msmcconnell, 2019-02-06:
@TakaHoribe 
I look forward to seeing what you come up with. I don't have a specific QP solver in mind yet, but a quick google search does present some open source projects.

TakaHoribe, 2019-03-18:
For now, we have checked qpOASES and pure-Eigen (just solve without constraint) for MPC optimization. Is there any suggestion for qp solver libraries?

closed 2019-06-04 by gbiggs