autowarefoundation/autoware #1971: Dose icp_matching node works well on the bag that you offered (sample_moriyama_150324.bag)

liuqi8827, 2019-02-10:
Hi 

Dose icp_matching node works well on the bag that you offered (sample_moriyama_150324.bag)?
I tested icp_matching node by running icp_matching.launch. The gps_pose in Rviz is right but base_link only moved at some times in Rviz, most of the time base_link did not move. It seems that the point cloud that Lidar collected dose not registrate with the map. 
I tested the bag by ndt_matching node using the same operating procedures, it worked well.
There is the error in Rviz when I run icp_matching.launch:
[image]

[image]

[image]

Thanks!

## Specifications

  - Ubuntu version:16.04
  - ROS version:Kinetic
  - Autoware branch:develop

YamatoAndo, 2019-02-12:
@liuqi8827 
Thanks for the report.
The computational cost of ICP Matching largely depends on the density of map points.
In my environment (i7-7700, RAM32GB, GTX-1080Ti) it takes about 5 seconds to calculate once.
The RViz error is because the ICP calculation is too slow and the proper TF can not be used.
Please down-sample the map with [PCD Filter] on the [Map] tab.
I tried to downsample the map to a 2.0 m voxel and it worked normally in about 50 ms per cycle.

We usually use NDT, so we do not maintain ICP frequently.
Sorry for the inconvenience.

liuqi8827, 2019-02-13:
@YamatoAndo 
Yes, it worked!

I down-sampled the map then it worked.

Thanks a lot.

closed 2019-02-14 by YamatoAndo