autowarefoundation/autoware #1973: Autoware build failed on Jetson TX2

duclinhfetel, 2019-02-11:
I use command:  ./catkin_make_release

# Bug
dw/core/Context.h: No such file or directory

[image]

## Specifications

  - Ubuntu version: 16.04
  - ROS version: kinetic - version: 1.12.14
  - Autoware branch: git clone https://github.com/CPFL/Autoware.git --recurse-submodules

Please,
How to solve this problem?
Thanks you for all support.

filiperinaldi, 2019-02-11:
The missing file (dw/core/Context.h) seems to be part of NVIDIA's SDK. Have you installed the driveworks SDK?

duclinhfetel, 2019-02-12:
> The missing file (dw/core/Context.h) seems to be part of NVIDIA's SDK. Have you installed the driveworks SDK?

Thanks,
Driveworks it's not support Jetson TX2. :(

naisy, 2019-02-15:
Hi @duclinhfetel,

On TX2, you need `CATKIN_BLACKLIST_PACKAGES`.
And use Autoware 1.9.1. (1.10.0 probably will fail to start.)

`bash -c 'source /opt/ros/kinetic/setup.bash; catkin_init_workspace; cd ../; ./catkin_make_release -DCATKIN_BLACKLIST_PACKAGES="autoware_driveworks_interface;autoware_driveworks_gmsl_interface" -j3'`

In order to run Autoware with TX2, consider using M.2 SSD from capacity and ROSBAG writing speed.

duclinhfetel, 2019-02-16:
> Hi @duclinhfetel,
>
> On TX2, you need `CATKIN_BLACKLIST_PACKAGES`.
> And use Autoware 1.9.1. (1.10.0 probably will fail to start.)
>
> `bash -c 'source /opt/ros/kinetic/setup.bash; catkin_init_workspace; [...]

Thanks,
I will try on the next week.

closed 2020-05-29 by gbiggs