autowarefoundation/autoware #1994: Build Autoware on PX2, and link Nvidia Driveworks API target, but Error for: dw/core/Context.h: No such file or directory

majichang, 2019-02-17:
[ 56%] Built target bond_generate_messages_cpp
Scanning dependencies of target nodelet_generate_messages_eus
[ 56%] Built target nodelet_generate_messages_eus
Scanning dependencies of target bond_generate_messages_eus
[ 56%] Built target bond_generate_messages_eus
Scanning dependencies of target autoware_driveworks_gmsl_interface_node
[ 56%] Building CXX object driveworks/packages/autoware_driveworks_gmsl_interface/CMakeFiles/autoware_driveworks_gmsl_interface_node.dir/src/gmsl_camera_multiple/gmsl_node.cpp.o
In file included from /home/nvidia/Autoware/ros/src/driveworks/packages/autoware_driveworks_gmsl_interface/include/SekonixGmslCamera.hpp:71:0,
                 from /home/nvidia/Autoware/ros/src/driveworks/packages/autoware_driveworks_gmsl_interface/src/gmsl_camera_multiple/gmsl_node.cpp:40:
/home/nvidia/Autoware/ros/src/driveworks/packages/autoware_driveworks_gmsl_interface/include/DriveWorksApi.hpp:66:29: fatal error: dw/core/Context.h: No such file or directory
compilation terminated.
driveworks/packages/autoware_driveworks_gmsl_interface/CMakeFiles/autoware_driveworks_gmsl_interface_node.dir/build.make:62: recipe for target '

majichang, 2019-02-17:
Can anyone else do a favor for me! looking for your award, thanks!

duclinhfetel, 2019-02-17:
Hi @majichang 

The missing file (dw/core/Context.h) seems to be part of NVIDIA's SDK. Have you installed the driveworks SDK?

majichang1990, 2019-02-17:
@duclinhfetel

yeah， I had successed install driveworks-0.3 on PX2，and the path is /usr/local/driveworks

filiperinaldi, 2019-02-18:
This is the same error on the recent issue #1973.
Perhaps FindDriveWorksSDK.cmake should fail earlier.

majichang, 2019-02-21:
@filiperinaldi Very thanks for you, I had deal With it. the Autoware Project had successed build on PX2 with  
 Nvidia DriveWorks.

jraclarke, 2019-02-21:
I still seem to be getting this problem. I did workaround it by explicitly putting the absolute driveworks include directory in CMakeLists.txt.  However, this threw "nvmedia_image.h: no such file or directory" instead though - this is missing in my driveworks sdk.  @majichang does nvmedia_image.h exist in your setup?

allenchen8210, 2019-05-24:
Hi @majichang  and @jraclarke  
I have same issue. I install autoware by this [page](https://github.com/autowarefoundation/autoware/blob/master/docs/en/installation_with_drivepx2.md).
I  install driveworks-0.6 on PX2 successfully.

Is there any suggestion ? Thank u

JWhitleyAStuff, 2019-05-28:
@majichang - Please see https://github.com/autowarefoundation/autoware/wiki/Support-guidelines for information on requesting support. This issue tracker is only for confirmed bugs and new features. Use [ROS Answers](https://answers.ros.org/questions/scope:all/sort:activity-desc/tags:autoware/page:1/) for general support questions.

closed 2020-05-29 by gbiggs