autowarefoundation/autoware #1995: NDT matching reliability

StanleyHusai, 2019-02-17:
In ndt-matching, there is a parameter named NDT_Reliability which is defined as: 
ndt_reliability.data = Wa * (exe_time / 100.0) * 100.0 + Wb * (iteration / 10.0) * 100.0 +
                           Wc * ((2.0 - trans_probability) / 2.0) * 100.0;
In my understanding, exe_time is the time of matching, iteration is the number of matching iteration, trans_probability is the probability score. Am i right?
So my question is What exactly is it？How can i use it?
For example is there any relationship between reliability and NDT-matching performance or NDT-matching positioning error?
Thanks so much for your answering!

kitsukawa, 2019-02-20:
@StanleyHusai As you said, exe_time is the time of matching, iteration is the number of matching iteration, trans_probability is the probability score. However, the correctness of the formula is not verified since this value is made for temporarily use(There is no topics subscribing this value). So please do not trust this value. We are working on quantification of the reliability of localization in a more robust way.

StanleyHusai, 2019-02-20:
Thank you so much!

yukkysaito, 2019-03-01:
https://github.com/CPFL/Autoware/issues/2025

closed 2019-03-04 by yukkysaito