autowarefoundation/autoware #1997: Add ament_cmake into build system

maximosipov, 2019-02-17:
# New Feature
In order to transition to ROS2 it is necessary to convert build system to ament_cmake and update packages according to [Migration Guide](https://index.ros.org/doc/ros2/Contributing/Migration-Guide/).

There are several build processes in Autoware to update and test:
- Travis CI is obsolete and replaced by Gitlab CI (will not be updated, need to eliminate .travis.yml and update docs)
- Gitlab CI is the current choice (will be updated based on https://github.com/esteve/Autoware/tree/feature/use_ament_vector_map, requires linking autoware repo into Gitlab from Github and updating docs)
- Native Docker build (will be updated on the top of #1982, however currently uses catkin due to #1853)
- Cross Docker builds (will be updated on the top of #1982, may be affected by #1853)
- Native host build (will be updated)

In addition, currently using ament_cmake requires using ament_ from ROS2 distribution which is not a recommended practice, see [Discourse Topic](https://discourse.ros.org/t/ros2-vs-ros1-or-more-like-ros2-ros1/7319/6?u=maximosipov), however it demonstrated to work.

gbiggs, 2019-02-18:
Is using Gitlab CI for a Github project a good idea? What benefits does Gitlab CI give us over Travis CI or another agnostic tool like CircleCI?

maximosipov, 2019-02-18:
@gbiggs Gitlab CI was introduced by #1563, I think it make sense to use just one framework, whatever the choice is, and the current support for Gitlab CI is definitely more flexible.

sgermanserrano, 2019-02-18:
@gbiggs Gitlab Ci was introduced as a replacement for Travis CI to overcome the problems that Travis CI had to run the unit tests. Also as a way to bring .Auto and .AI closer

esteve, 2019-02-18:
@gbiggs like @sgermanserrano said, we wanted to bring the two projects closer and as a first step we decided to use GitLab, also we were having lots of issues with Travis, so we needed to switch to something else anyway.

Like CircleCI, GitLab runners can be hosted on premises, however and although we'll initially only support Linux, GitLab supports macOS and Windows as well (e.g. @maximosipov reported a bug in Authoware.Auto when building on macOS, so we want to support that too)

I personally like Azure Pipelines a lot, but in order to keep the infrastructure consolidated, we went with GitLab.

gbiggs, 2019-02-19:
Yeah, I'm a fan of Azure Pipelines, too.

Thanks for the explanations! No one's been able to tell me for four weeks. :laughing:

esteve, 2019-02-19:
@gbiggs sorry about that, we should have a document explaining this sort of decisions. Moreover, we'll also have to look into consolidating the CI scripts as well, right now we use https://github.com/ros-industrial/industrial_ci, and custom scripts in GitLab and Travis. `industrial_ci` is a great solution, however it doesn't support `colcon` nor ROS2 yet, so we'll have to either contribute it upstream or keep using our scripts. I'd rather push upstream as much as possible, instead maintaining our own stuff, but I don't have an estimate of how much work it'd require to support `colcon` in `industrial_ci`.

130s, 2019-02-19:
Thanks for the mention for industrial_ci (I'm one of the maintainer). ROS2/Colcon support is discussed in https://github.com/ros-industrial/industrial_ci/issues/333 probably because of the lack of concrete drive. If you could voice up there that could help someone to pick up the pieces :)

gbiggs, 2019-02-19:
I'm all in favour of pushing anything we can upstream. It's good for the community, and it means less for us to maintain all by ourselves.

esteve, 2019-02-20:
@130s thanks for dropping by. I followed that ticket a bit, and the consensus seemed that `industrial_ci` would be rewritten in Python first before adding support for `colcon`. Anyway, I've responded there to get the ball rolling.

closed 2020-05-29 by gbiggs