autowarefoundation/autoware #1998: ndt_matching error when using pcl_anh_gpu on DrivePX2

phoebus04, 2019-02-18:
# Bug
When using ndt_matching by Method Type pcl_anh_gpu,  error "out of memory" happened
## Expected Behavior
ndt_matching by GPU works

## Actual Behavior
ndt_matching by GPU doesnot work

## Steps to Reproduce the Problem

  1. select pcl_anh_gpu in ndt_matching app on DrivePX2

## Specifications

  - Ubuntu version:16.04
  - ROS version:kinetic
  - Autoware branch:

SUMMARY
========

PARAMETERS
* /ndt_matching/get_height: False
* /ndt_matching/imu_topic: /imu_raw
* /ndt_matching/imu_upside_down: False
* /ndt_matching/method_type: 2
* /ndt_matching/offset: linear
* /ndt_matching/output_log_data: False
* /ndt_matching/queue_size: 1
* /ndt_matching/use_gnss: 1
* /ndt_matching/use_imu: False
* /ndt_matching/use_local_transform: False
* /ndt_matching/use_odom: False
* /rosdistro: kinetic
* /rosversion: 1.12.14

NODES
/
ndt_matching (lidar_localizer/ndt_matching)

ROS_MASTER_URI=http://parker-a:11311

nvrm_gpu: Bug 200215060 workaround enabled.
Error: out of memory /home/nvidia/Autoware/ros/src/computing/perception/localization/lib/ndt_gpu/src/SymmetricEigenSolver.cu 10
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'
what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
[ndt_matching-1] process has died [pid 8159, exit code -6, cmd /home/nvidia/Autoware/ros/devel/lib/lidar_localizer/ndt_matching __name:=ndt_matching __log:=/home/nvidia/.ros/log/b7fb276a-3342-11e9-bbdf-00044bc4763b/ndt_matching-1.log].
log file: /home/nvidia/.ros/log/b7fb276a-3342-11e9-bbdf-00044bc4763b/ndt_matching-1*.log
process[ndt_matching-1]: started with pid [8159]
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

yukkysaito, 2019-03-01:
ndt_anh _cpu or gpu consumes a lot of memory due to the tree structure relationship(octree).
It can only read about 300 m of map

phoebus04, 2019-03-04:
Thanks. I tryed to set the Resolution to 4, and then it works normally. so what does Resolution option mean here?

closed 2020-05-29 by gbiggs