autowarefoundation/autoware #2002: Quick Start Error

jjimin, 2019-02-19:
I've just followed the quick start but it isn't working. I don't know what should I do.
Could anyone please help me solve this. Thank you.

## Actual Behavior
(Displays in RViz)
In Points Raw.
For frame [velodyne]: Frame [velodyne] does not exist
In Vehicle Model,
Parameter [robot_description] does not exist, and was not found by searchParam()
In Occupancy Grid Map,
No map received

[image]

[image]

## Steps to Reproduce the Problem

  1. Compile using `./catkin_make_release`
  1. Execute `./run`
  1. Check that the use_sim_time is 'false'
  1. Go to the Simulation tab of runtime manager, and load `/home/user/.autoware/sample_moriyama_150324.bag`
  1. Play the loaded data, and pause it
  1. Go to the QuickStart tab and load `/home/user/.autoware/my_launch/my_map.launch` and `/home/user/.autoware/my_launch/my_localization.launch`
  1. Launch rviz and open the configuration file `/home/user/Autoware/ros/src/.config/rviz/default.rviz`

## Specifications

  - Ubuntu version: 16.04
  - ROS version: kinetic

4artit, 2019-02-21:
Which autoware version are you using?

andoh104, 2019-04-07:
@jjimin 

After v.1.7 my_launch files are in autoware/docs/quick_start . Did you use it ? (you don't have to copy it)
I think some messages were in your terminal when autoware was launched.

YangLiu14, 2019-04-27:
Hi, I have the exact same issue, 
I am using docker(case 1), but with nvidia-docker2.  I am using the latest version of Autoware from master-branch. And I did the exact steps following the demo tutorial

andoh104, 2019-04-27:
Are you using these launch file ?
https://github.com/autowarefoundation/autoware/tree/master/docs/quick_start

YangLiu14, 2019-04-28:
> Are you using these launch file ?
> https://github.com/autowarefoundation/autoware/tree/master/docs/quick_start

Hi, thank you for the reply, yes, I am using exactly that, and I am following each step according to the Demo tutorial. https://github.com/autowarefoundation/autoware/wiki/Demo

YangLiu14, 2019-04-28:
And it's not only the `my_map.launch` that I have some problem with. Opening the following launch files from **_/home/autoware/Autoware/docs/quick_start_** produce the following errors:

 `my_sensing.launch`

['roslaunch', '/home/autoware/Autoware/docs/quick_start/my_sensing.launch']
pid=4371
sched policy=OTHER prio=0
while processing /opt/ros/kinetic/share/velodyne_pointcloud/launch/velodyne_hdl32e.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/opt/ros/kinetic/share/velodyne_pointcloud/launch/velodyne_hdl32e.launch'
The traceback for the exception was written to the log file

`my_localization.launch`

while processing /home/autoware/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/setup_tf.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/autoware/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/setup_tf.launch'
The traceback for the exception was written to the log file

`my_detection.launch`

while processing /home/autoware/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/calibration_publisher.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/autoware/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/calibration_publisher.launch'
The traceback for the exception was written to the log file

`my_mission_planning.launch`

[config_waypoint_loader_rostopic-4] process has died [pid 4780, exit code 1, cmd /opt/ros/kinetic/lib/rostopic/rostopic pub -l /config/waypoint_loader autoware_msgs/ConfigWaypointLoader {multi_lane_csv: /home/autoware/.autoware/data/path/moriyama_path.txt, replanning_mode: false, velocity_max: 20.0, velocity_min: 4.0, accel_limit: 0.98, decel_limit: 0.98, radius_thresh: 20.0, radius_min: 6.0, resample_mode: true, resample_interval: 1.0, velocity_offset: 4, end_point_offset: 5} __name:=config_waypoint_loader_rostopic __log:=/home/autoware/.ros/log/d6bc713a-69cb-11e9-af6a-000ec6d9e5c8/config_waypoint_loader_rostopic-4.log].
log file: /home/autoware/.ros/log/d6bc713a-69cb-11e9-af6a-000ec6d9e5c8/config_waypoint_loader_rostopic-4*.log

`my_motion_planning.launch`

while processing /home/autoware/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/vehicle_socket.launch:
Invalid roslaunch XML syntax: [Errno 2] No such file or directory: u'/home/autoware/Autoware/ros/install/runtime_manager/share/runtime_manager/scripts/vehicle_socket.launch

gbiggs, 2019-04-29:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

closed 2019-04-29 by gbiggs

YangLiu14, 2019-05-05:
This is my preliminary solution for the problem, if anyone is interested see this post. 
https://discourse.ros.org/t/autoware-1-11-0-quick-start-demo-failed-to-load-the-map/9019