autowarefoundation/autoware #2010: How to run a lane change in new Runtime Manager

AthpapAutowareUser, 2019-02-21:
In the new Runtime Manager manager if  want to run a lane change simulation there some option which are missing like : dp_planner ff_waypoint_follower. 

https://www.youtube.com/watch?v=O0OtSy9pS7M

Which are the new buttons in the latest version ?

hatem-darweesh, 2019-02-22:
hello,
- you can use older version  of autoware. no dramatic changes with dp_planner and ff_waypoint_follower in the new version. 
- these nodes are still in the current latest version , you can launch the nodes from command line.

AthpapAutowareUser, 2019-02-22:
Thank you for the direct response. 
Would it be possible to type the me steps in order to succeed that in the version I have  ???

(e.g. what are the commands, the order ot actions etc )

closed 2020-05-29 by gbiggs