autowarefoundation/autoware #2022: Amentification meta-ticket

gbiggs, 2019-02-25:
This ticket is to track the amentification of packages. Once all packages are done, we can close this ticket and consider the feature complete.

- [ ] [Use ament for autoware_build_flags](https://github.com/CPFL/Autoware/pull/2014)
- [ ] [Integrate ament into CI and docker builds](https://github.com/CPFL/Autoware/pull/2007)

maximosipov, 2019-02-26:
amathutils_lib and decision_maker:
https://github.com/CPFL/Autoware/pull/2016

maximosipov, 2019-02-26:
diag_lib and fake_autoware_nodes:
#2033

maximosipov, 2019-02-26:
state_machine_lib:
#2034

maximosipov, 2019-02-26:
libvectormap and trafficlight_recognizer:
#2035

maximosipov, 2019-03-01:
pixel_cloud_fusion and range_vision_fusion:
#2074

maximosipov, 2019-03-07:
idar_euclidean_cluster_detect, lidar_fake_perception and lidar_naive_l_shape_detect:
#2099

closed 2020-05-29 by gbiggs