autowarefoundation/autoware #2025: NDT mapping determinism and accuracy criteria

TBondat, 2019-02-25:
I have two question marks about NDT algorithm used in autoware:
- is NDT implementation deterministic? So by giving always the same point clouds and init_guess to ndt.align, is it 100% sure that the output will be the same?
I got some differences in results depending on the conditions i use ndt, so i wanted to make sure that normally we should have deterministic NDT.

- As i get some z-drift with ndt, I try to trace the ndt accuracy. Is fitness_score a robust quality criteria? i saw in a recent question that transform_probability should not be used. so is it the same for fitness_score? What other variable could be used if fitness is not good?

thanks!

yukkysaito, 2019-03-01:
* is NDT implementation deterministic?
Since the convergence degree varies depending on the initial value, it is not definitive. The initial value also varies depending on the execution time

* Is fitness_score a robust quality criteria?
It is a very difficult problem to decide the numerical value representing the quality of ndt. It is necessary to decide at what time the quality is bad or good. However, ndt often becomes unstable when the yaw rate is high, and fitness score seems to be consistent with instability.
He is trying to introduce new rating values (@YamatoAndo )

TBondat, 2019-03-02:
thanks for the answers. When you say the initial value, do you mean the initial guess provided to ndt or do you talk about internal ndt parameter initialization (in this case, where is it defined?) ?

I agree it seems complex to get proper metrics about map quality, and fitness score is not enough. Nice to hear that some work is done on this part! @YamatoAndo  can we have some first hints about the future metrics that we will get in a futur release? :)

yukkysaito, 2019-03-04:
>> When you say the initial value, do you mean the initial guess provided to ndt or do you talk about internal ndt parameter initialization (in this case, where is it defined?) ?

It mean the initial guess provided to ndt. (not parameter)

StanleyHusai, 2019-06-27:
@yukkysaito @YamatoAndo Thanks for your answer! Do you have new results about the rating values of NDT matching?

amc-nu, 2019-06-27:
Autoware has moved from GitHub to [GitLab](https://gitlab.com/autowarefoundation/autoware.ai). 

IF YOU ARE MAKING A SUPPORT REQUEST, DO NOT POST AN ISSUE.
Follow the [support guidelines](https://gitlab.com/autowarefoundation/autoware.ai/autoware/wikis/Support-guidelines) and ask your question at [ROS Answers](https://answers.ros.org/questions/ask/?tags=autoware).

closed 2019-06-27 by amc-nu