autowarefoundation/autoware #2041: VoxelNet, LMNet detection

JungHY, 2019-02-27:
this is over view of Autoware

https://github.com/CPFL/Autoware/wiki/Overview

in Detection section of Computing/Perception there is a explanation about lidar detector

it says that Autoware provides DNN-based algorithms such as VoxelNet and LMNet

but i can't find any of these algorithms in master or other branches

where can i find these algorithms? or guidance to add new algorithm or pakcage

thank you

yukkysaito, 2019-03-01:
@amc-nu @cirpue49 Do you know anything?

cirpue49, 2019-03-01:
Hi, if you are looking for CNN based detection algorithm, please check this one. https://github.com/CPFL/Autoware/pull/2029

Thanks.

closed 2019-03-01 by cirpue49

Shawn-Zhuang, 2020-08-03:
@k0suke-murakami Hi, I am also interested in LMNet, which can run without GPU. 
 [#2029](https://github.com/Autoware-AI/autoware.ai/pull/2029) is the algorithm of PointPillars instead of LMNet. 
I want to test the result in a computer without GPU, so I still need to know how to test with LMNet.
Is the function LMNet removed from Autoware? I can't find any instruction to use it.