autowarefoundation/autoware #2052: [Feature] Add Carla Simulator Bridge

hakuturu583, 2019-03-01:
Carla is a good autonomous driving simulator and it has a lot of functions and there are already exist carla-autoware repo.(https://github.com/carla-simulator/carla-autoware)

# New Feature
I want to integrate this repository code into Autoware and integrate carla-autoware bridge into Autoware.

sgermanserrano, 2019-03-05:
@hakuturu583 what does this feature need to add to the Autoware repo?

hakuturu583, 2019-03-05:
@sgermanserrano I want to add carla-autoware bridge docker image to the Autoware.

esteve, 2019-03-05:
@hakuturu583 Yes, but how? As a submodule? Build Carla in a Dockerfile and ship it with Autoware? We need more information and a concrete plan for every ticket.

hakuturu583, 2019-03-07:
@esteve Now, I am thinking Build Carla and Carla-Autoware bridge Dockerfile in the Autoware.
And, enable launch these docker image with roslaunch command.

sgermanserrano, 2019-03-07:
@hakuturu583 what would be different from the current docker implementation in CARLA? Can it just be added as a submodule?

hakuturu583, 2019-03-07:
@sgermanserrano I am planning to use current docker implementation in CARLA and autoware-carla bridge Docker image.
This is because now carla-bridge setup is very difficult and I thought it is unable to solve it by using rosdep.
So, I think it is better for us to add carla-bridge as a docker image.

sgermanserrano, 2019-03-08:
@hakuturu583 what's is the work proposed in this feature then? adding it as a submodule?

hakuturu583, 2019-03-19:
@sgermanserrano Now, I am planning to add thease feature as Docker image, not submodule.
Now, carla bridge setup is too much complicated...
So, I think it is the best way to add this feature as docker image.

hakuturu583, 2019-03-19:
And, I also sending request to the carla people makes it easier to run with Autoware.
(distribute bridge as python or ROS package etc....)
If they do so, I want to change the implementation of the feature.

sgermanserrano, 2019-03-19:
> @sgermanserrano Now, I am planning to add thease feature as Docker image, not submodule.
> Now, carla bridge setup is too much complicated...
> So, I think it is the best way to add this feature as docker image.

@hakuturu583 doing so increases the amount of code/files that we need to maintain, it would be better to PR the improvements to the Carla repository since they already have some implementation of the bridge

esteve, 2019-03-19:
I found https://github.com/carla-simulator/carla/issues/81 which if addressed, would basically solve all the problems. @hakuturu583 can you chime in there? Making the Carla Python library installable via PIP would just solve all the issues you had.

hakuturu583, 2019-03-20:
@esteve Good. I think if the Carla Python library can be installed via PIP, then we can resolve dependency via rosdep and it makes Carla and Autoware much easily.

vimalrajliya, 2019-11-19:
Any updates on this?

closed 2020-05-29 by gbiggs