autowarefoundation/autoware #2087: Why the Vector Map Builder didn't work like the introduction video?

MouriNaruto, 2019-03-05:
I used that tool. But I found that when I imported my PCD file, it didn't work like that video.

Tool: https://tools.tier4.jp/feature/vector_map_builder/

Video
[image]

Mine
[image]

I want to know how to solve the problem. Thanks.

Mouri.

gbiggs, 2019-03-06:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

closed 2019-03-06 by gbiggs

hroachewilson, 2019-05-22:
fwiw, using a tool like pcl_convert_pcd_ascii_binary to convert the pcl file to binary or ascii solves the issue.