autowarefoundation/autoware #2093: [Bug] ring ground filter runtime manager parameters

javcursor, 2019-03-06:
Hi!

When you modify the ring_ground_filter parameters in runtime manager they aren't connected and it use the configured values in ring_ground_filter.launch.

[image]

[image]

Specifications

    Ubuntu version:16.04
    ROS version:kinetic
    Autoware branch: autoware 1.9.1 & autoware 1.10

amc-nu, 2019-03-11:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

gbiggs, 2019-03-11:
Actually, it could be more detailed but this one *is* a bug report.

closed 2020-05-29 by gbiggs