autowarefoundation/autoware #2094: ndt_gpu not built using colcon_release on ros kinetic gpu desktop

manikantarav, 2019-03-06:
I git cloned the autoware recent release. at last step, after ./calcon_release, ndt_gpu package could not built, giving error. i posted log here. 
# Bug

## Expected Behavior

Autoware should have been bulit successfully.

## Actual Behavior
ptxas info    : Used 56 registers, 144 bytes smem, 448 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu20computePointHessian0EPfS0_S0_iPiiPdS2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu20computePointHessian0EPfS0_S0_iPiiPdS2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_S2_
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 56 registers, 144 bytes smem, 496 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu22computePointGradients1EPfS0_S0_iPiiPdS2_S2_S2_S2_' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu22computePointGradients1EPfS0_S0_iPiiPdS2_S2_S2_S2_
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 40 registers, 96 bytes smem, 408 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu22computePointGradients0EPfS0_S0_iPiiPdS2_S2_S2_S2_S2_S2_S2_' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu22computePointGradients0EPfS0_S0_iPiiPdS2_S2_S2_S2_S2_S2_S2_
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 40 registers, 96 bytes smem, 432 bytes cmem[0]
/usr/include/pcl-1.7/pcl/point_cloud.h:586:100: error: template-id ‘getMapping<PointT>’ used as a declarator
       friend boost::shared_ptr<MsgFieldMap>& detail::getMapping<PointT>(pcl::PointCloud<PointT> &p);
                                                                                                    ^
/usr/include/pcl-1.7/pcl/point_cloud.h:586:100: error: ‘getMapping’ is neither function nor member function; cannot be declared friend
cc1plus: error: expected ‘;’ at end of member declaration
/usr/include/pcl-1.7/pcl/point_cloud.h:586:111: error: expected ‘)’ before ‘&’ token
CMake Error at ndt_gpu_generated_NormalDistributionsTransform.cu.o.cmake:266 (message):
  Error generating file
  /home/kpit/manikanta/Autoware/ros/build/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/./ndt_gpu_generated_NormalDistributionsTransform.cu.o

make[2]: *** [CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o] Error 1
CMakeFiles/ndt_gpu.dir/build.make:77: recipe for target 'CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o' failed
CMakeFiles/Makefile2:259: recipe for target 'CMakeFiles/ndt_gpu.dir/all' failed
make[1]: *** [CMakeFiles/ndt_gpu.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

## Steps to Reproduce the Problem

  1.clone Autoware
  2. ./colcon_release 

## Specifications

  - Ubuntu version: 16.04
  - ROS version: ROS kinetic
  - Autoware branch: 1.10.0
  - NVIDIA Drivers: Driver Version: 418.43       CUDA Version: 10.1
  - GPU: Geforce RTX 2080
[streams.log](https://github.com/CPFL/Autoware/files/2936107/streams.log)
[command.log](https://github.com/CPFL/Autoware/files/2936109/command.log)
[stdout_stderr.log](https://github.com/CPFL/Autoware/files/2936111/stdout_stderr.log)

manikantarav, 2019-03-06:
Now i build using catkin_make_release. But same error. Here i shared the logs. please check and 

***************************************************************************************************
[ 42%] Generating Javascript code from autoware_msgs/ImageRect.msg
[ 42%] Generating Lisp code from vector_map_msgs/SideWalk.msg
[ 42%] Generating Javascript code from vector_map_msgs/PointArray.msg
[ 42%] Generating Python from MSG vector_map_msgs/RoadMark
CMake Error at ndt_gpu_generated_NormalDistributionsTransform.cu.o.cmake:266 (message):
  Error generating file
  /home/kpit/manikanta/Autoware/ros/build/computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/./ndt_gpu_generated_NormalDistributionsTransform.cu.o

[ 42%] Generating EusLisp code from vector_map_msgs/Point.msg
computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/build.make:77: recipe for target 'computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o' failed
make[2]: *** [computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o] Error 1
CMakeFiles/Makefile2:2181: recipe for target 'computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/all' failed
make[1]: *** [computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 42%] Generating C++ code from vector_map_msgs/StreetLight.msg
[ 42%] Generating C++ code from vector_map_msgs/FenceArray.msg
[ 42%] Generating C++ code from autoware_msgs/BrakeCmd.msg
[ 42%] Generating C++ code from vector_map_msgs/StreetLightArray.msg
[ 42%] Generating C++ code from autoware_msgs/ICPStat.msg
[ 42%] Generating C++ code from autoware_msgs/Signals.msg
[ 42%] Generating C++ code from autoware_msgs/AdjustXY.msg
[ 42%] Generating C++ code from autoware_msgs/StateCmd.msg
[ 42%] Generating Javascript code from autoware_msgs/DetectedObject.msg
[ 42%] Generating Javascript code from vector_map_msgs/CrossRoad.msg
[ 42%] Generating Lisp code from vector_map_msgs/WhiteLineArray.msg
[ 42%] Generating Python from MSG vector_map_msgs/CrossWalk
[ 42%] Generating EusLisp code from vector_map_msgs/Pole.msg
[ 42%] Generating Javascript code from vector_map_msgs/WhiteLine.msg
[ 42%] Generating C++ code from vector_map_msgs/RailCrossing.msg
[ 42%] Generating Lisp code from vector_map_msgs/Curb.msg
[ 42%] Generating Javascript code from autoware_msgs/VscanTracked.msg
[ 42%] Building CXX object sensing/drivers/lidar/packages/sick/ldmrs/sick_ldmrs_driver/src/driver/src/CMakeFiles/LDMRS_Example.dir/main.cpp.o
[ 42%] Linking CXX shared library /home/kpit/manikanta/Autoware/ros/devel/lib/libsick_ldmrs.so
[ 42%] Generating EusLisp code from vector_map_msgs/PointArray.msg
[ 42%] Generating EusLisp code from vector_map_msgs/CrossRoad.msg
[ 42%] Generating Javascript code from vector_map_msgs/WallArray.msg
[ 42%] Generating C++ code from vector_map_msgs/SideWalk.msg
*********************************************************************************************************
[CMakeError.log](https://github.com/CPFL/Autoware/files/2936598/CMakeError.log)
[CMakeOutput.log](https://github.com/CPFL/Autoware/files/2936599/CMakeOutput.log)

sgermanserrano, 2019-03-06:
we are currently resolving issues that arise with CUDA 10 (@amc-nu ) you need to use CUDA 9 for now

amc-nu, 2019-03-06:
This is caused by your CUDA version. Downgrade to cuda 9.

manikantarav, 2019-03-07:
[CMakeError.log](https://github.com/CPFL/Autoware/files/2940059/CMakeError.log)
[CMakeOutput.log](https://github.com/CPFL/Autoware/files/2940060/CMakeOutput.log)

Thank you for quick reply. now i Downgraded it to CUDA 9. still couldnot build. can you check once. @amc-nu @sgermanserrano 

**********************************************************************************

[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/adi_driver/adis16470_node
[  1%] Built target _lgsvl_msgs_generate_messages_check_deps_Detection2D
[  1%] Built target adxl345_node
Scanning dependencies of target cuda_binary
Scanning dependencies of target _autoware_can_msgs_generate_messages_check_deps_CANData
Scanning dependencies of target _autoware_can_msgs_generate_messages_check_deps_CANPacket
[  1%] Generating GPU_function.cubin
nvcc fatal   : Value 'sm_75' is not defined for option 'gpu-architecture'
computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/cuda_binary.dir/build.make:60: recipe for target 'computing/perception/detection/vision_detector/libs/dpm_ttic/GPU_function.cubin' failed
make[2]: *** [computing/perception/detection/vision_detector/libs/dpm_ttic/GPU_function.cubin] Error 1
CMakeFiles/Makefile2:15712: recipe for target 'computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/cuda_binary.dir/all' failed
make[1]: *** [computing/perception/detection/vision_detector/libs/dpm_ttic/CMakeFiles/cuda_binary.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Built target _autoware_can_msgs_generate_messages_check_deps_CANData
[  1%] Built target _autoware_can_msgs_generate_messages_check_deps_CANPacket
[  1%] Built target adis16470_node
[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/hexacam/hexacam_node
[  1%] Built target hexacam_node
[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/fake_drivers/fake_camera
[  1%] Linking CXX shared library /home/kpit/manikanta/Autoware/ros/devel/lib/libfastvirtualscan.so
[  1%] Built target fastvirtualscan
[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/vectacam/vectacam_node
[  1%] Built target fake_camera
[  1%] Built target vectacam_node
ptxas info    : 0 bytes gmem
ptxas info    : Compiling entry function '_ZN3gpu12convertInputIN3pcl8PointXYZEEEvPT_PfS5_S5_i' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu12convertInputIN3pcl8PointXYZEEEvPT_PfS5_S5_i
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 14 registers, 356 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu12convertInputIN3pcl9PointXYZIEEEvPT_PfS5_S5_i' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu12convertInputIN3pcl9PointXYZIEEEvPT_PfS5_S5_i
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 14 registers, 356 bytes cmem[0]
[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/image_processor/image_rotator
[  1%] Built target image_rotator
[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/laser_scan_converter/laser_scan_converter_node
[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/image_processor/image_rectifier
[  1%] Built target laser_scan_converter_node
[  1%] Built target image_rectifier
[  1%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/garmin/garmin_gps_18x_lvc
[  1%] Built target garmin_gps_18x_lvc
[  2%] Linking CXX executable /home/kpit/manikanta/Autoware/ros/devel/lib/hokuyo/hokuyo_3d
[  2%] Built target hokuyo_3d
[  2%] Linking CXX shared library /home/kpit/manikanta/Autoware/ros/devel/lib/libndt_cpu.so
[  2%] Built target ndt_cpu
ptxas info    : 0 bytes gmem
ptxas info    : Compiling entry function '_ZN3gpu6gpuSumIiEEvPT_ii' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu6gpuSumIiEEvPT_ii
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 10 registers, 336 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu6gpuSumIdEEvPT_ii' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu6gpuSumIdEEvPT_ii
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 10 registers, 336 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu12gpuTransformEPfS0_S0_S0_S0_S0_iNS_12MatrixDeviceE' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu12gpuTransformEPfS0_S0_S0_S0_S0_iNS_12MatrixDeviceE
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 40 registers, 400 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu8sumScoreEPdii' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu8sumScoreEPdii
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 12 registers, 336 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu9matrixSumEPdiiiii' for 'sm_52'
[... 38 lines cut ...]
ptxas info    : Used 32 registers, 72 bytes smem, 400 bytes cmem[0]
ptxas info    : Compiling entry function '_ZN3gpu20computePointHessian1EPfS0_S0_iPiiPdS2_S2_S2_S2_S2_S2_S2_S2_S2_' for 'sm_52'
ptxas info    : Function properties for _ZN3gpu20computePointHessian1EPfS0_S0_iPiiPdS2_S2_S2_S2_S2_S2_S2_S2_S2_
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 56 registers, 144 bytes smem, 448 bytes cmem[0]
[... cut ...]
    0 bytes stack frame, 0 bytes spill stores, 0 bytes spill loads
ptxas info    : Used 40 registers, 96 bytes smem, 432 bytes cmem[0]
[  2%] Linking CXX shared library /home/kpit/manikanta/Autoware/ros/devel/lib/libpcl_omp_registration.so
[  2%] Built target pcl_omp_registration
Scanning dependencies of target ndt_gpu
[  2%] Linking CXX shared library /home/kpit/manikanta/Autoware/ros/devel/lib/libndt_gpu.so
[  2%] Built target ndt_gpu
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j20 -l20" failed

amc-nu, 2019-03-07:
@manikantarav Your GPU architecture (`sm_75`) RTX20x0, is not supported by CUDA 9.0. In this case you'll need to use CUDA 10.0. However, we will support it until the release of #2032 

In the meantime, probably you can try to use `nvidia-docker`. Please have a look at the docker version of Autoware. I don't have an RTX to test, so I can'T be sure if the virtualization might work, if you can test and report us back that would be awesome.

manikantarav, 2019-03-08:
@amc-nu now i switched to Cuda 10, and compiled using catkin_make_release. Now everything working fine.

About docker, i tried it one week back. i didnt save any logs. But rviz was not working that time, then i switched to source installation.

Thank you for help anyway.

amc-nu, 2019-03-12:
@manikantarav very probably you were using nvidia-docker 2, the docker image for Autoware <= v1.10 requires nvidia-docker v1. #1946 will bring support for it on 1.11.

Thanks for reporting back the result.

closed 2019-03-12 by amc-nu

leon0417, 2019-03-30:
@manikantarav I met the same problem, and now, I use RTX 2080ti. Please tell me which branch you used in Autoware, and which version for the nvidia driver and CUDA.

gbiggs, 2019-03-31:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

valgur, 2019-12-03:
I also encountered this bug in melodic with PCL 1.8 and CUDA 10.1. It's not ideal, but I found that as a temporary workaround you can simply comment out / remove the 
```cpp
friend boost::shared_ptr& detail::getMapping(pcl::PointCloud &p);
```
line and make `_mapping` public in `pcl/point_cloud.h`.