autowarefoundation/autoware #2097: [bug] Autoware on Melodic. fatal error: gazebo_plugins/gazebo_ros_block_laser.h: No such file or directory

amc-nu, 2019-03-07:
# Bug

### Error:
```
gazebo_ros_block_laser.cpp:26:10: fatal error: gazebo_plugins/gazebo_ros_block_laser.h: No such file or directory
 #include <gazebo_plugins/gazebo_ros_block_laser.h>
```
https://travis-ci.org/CPFL/Autoware/jobs/502609548#L2581

## Specifications

  - Ubuntu version: 18.04
  - ROS version: Melodic
  - Autoware branch: feature/melodic

Related to #2032 

# Solution
Add `gazebo_plugins` to the dependency list in `package.xml` in `Autoware/ros/src/simulation/gazebo_simulator/worlds/external/car_demo/car_demo/package.xml` 

https://github.com/CPFL/Autoware/tree/develop/ros/src/simulation/gazebo_simulator/worlds/external

yukkysaito, 2019-03-07:
Thank you. 
Please wait a bit because it will check in ubuntu 18.04 environment

amc-nu, 2019-03-23:
Thanks @yukkysaito 
fixed on latest melodic sync and latest develop
https://travis-ci.org/CPFL/Autoware/jobs/510224481

closed 2019-03-23 by amc-nu