autowarefoundation/autoware #2098: [bug] Autoware on Melodic. CUDA 10.0. Eigen::half: has no member 'x'

amc-nu, 2019-03-07:
# Bug

```
Eigen/src/Core/arch/CUDA/Half.h error: Eigen::half: has no member 'x'
```

## Specifications

  - Ubuntu version: 18.04
  - ROS version: Melodic
  - Autoware branch: feature/melodic

Related to #2032 #2094 

## Solution

Upgrade Eigen to at least 3.3.5. 

http://eigen.tuxfamily.org/index.php?title=ChangeLog#Eigen_3.3.5

Successfully tested on latest `3.3.7`.

JWhitleyAStuff, 2019-05-10:
More info on this bug: The specific commit that fixes the problem is here: https://bitbucket.org/eigen/eigen/commits/522a5b96727a7322a0daaad926b52157cb76b668?at=3.3

It was not included until **3.3.6** so you can use that version or above. I created a script that allows you to install this easily and non-destructively:

https://gist.github.com/JWhitleyAStuff/c4bc47a5ba7903b48bd2fab108fe2afe

JWhitleyAStuff, 2019-05-22:
Fixed by #2265 (though #2271 will have a better implementation).

gbiggs, 2019-05-23:
@JWhitleyAStuff Does that mean we can close this?

JWhitleyAStuff, 2019-05-23:
Yes. Can be closed.

closed 2019-05-24 by gbiggs