autowarefoundation/autoware #2111: [feature] a simple local path planner for obstacle avoidance

TakaHoribe, 2019-03-11:
A local path planner for obstacle avoidance.

# New Feature
The current planning architecture in Autoware is as follows.
 - global planner (waypoint loader)
 - freespace planner (aster_planner)
 - velocity planner (velocity_set)

Currently it is `aster_planner` that is responsible for the obstacle avoidance function, but the A* is a free-space planning algorithm (and thus, it's time consuming), so it should be only for a complicated situation. 
For an easy situation (such as just one obstacle), it would be good to employ a local path planner.
The function of the local path planner can be simple for practical use, such as referring to the cost map and selecting one from pre-calculated multiple path. 

There are some local planners in Autoware, but they have not been integrated or properly modularized for the above planning architecture. 

It is necessary to design a new planner or integrate the existing planner with substantial changes.

TakaHoribe, 2019-03-11:
A picture of the current architecture is in this issue.
https://github.com/CPFL/Autoware/issues/1677

One possibility is to include local path planner after `avoid_planner` (a*).

kennedywai, 2019-03-12:
I thought there was another feature called astar_avoid doing the job?
https://github.com/CPFL/Autoware/tree/feature/improve_astar_planner

aohsato, 2019-03-14:
@TakaHoribe Could you describe your "simple local path planner" with a figure or screenshot of your prototype?

closed 2020-05-29 by gbiggs