autowarefoundation/autoware #2115: [Feature] upgrade gmsl camera driver to DriveWorks 1.5 on NVIDIA Drive Platform

twbabyduck, 2019-03-12:
## Status
I would like to do a pull request; however, I am not quite understand the contributing guidelines and the workflow of the process. Does anyone could guide me through this?

## Specification

  - Ubuntu version : 16.04.x
  - ROS version : Kinetic Kame
  - DriveWorks version : 1.5+
  - Target Platform : NVIDIA DRIVE AGX Xavier, NVIDIA DRIVE AGX Pegasus

# New Feature
Upgrade `autoware_driveworks_gmsl_interface` && `autoware_driveworks_interface` to the latest DriveWorks API version (e.g. 1.5) 

### Future Release

- remove all the deprecated API, related files and references
- support up to 16 cameras (4 camera ports) for DRIVE™ AGX Developer System

sgermanserrano, 2019-03-13:
@twbabyduck please have a look [here](https://github.com/CPFL/Autoware/wiki/Contributing-to-Autoware) for the Contribution Guidelines.
@amc-nu can you have a look at the proposed feature?

amc-nu, 2019-03-23:
@sgermanserrano @twbabyduck Sure, I can have a look.

Leonardo092, 2019-04-10:
@twbabyduck I looked for your job but I could not find anything. Didn't you make it public on your repo?

twbabyduck, 2019-04-11:
@Leonardo092 
`sgermanserrano` said that I should wait until autoware release 1.11 to send PR, and I might commit my code next week due to heavy workload in recent days. 

@amc-nu 
Could you guide me through this for more hints on sending PR to autoware repo, and I would like to add some function to this gmsl camera driver's package. Do you think we should discuss before I send any requests?

jokla, 2019-04-11:
@twbabyduck Have you tested the node on the docker or outside? Thanks!

gbiggs, 2019-04-14:
The [contribution guidelines](https://github.com/autowarefoundation/autoware/wiki/Contributing-to-Autoware) give a detailed description of the PR process. If there is anything specific you do not understand in there, then please ask, but you need to tell us what exactly you do not understand.

jokla, 2019-04-15:
@twbabyduck in case you are ready to share the drivers, I would be happy to test it.

twbabyduck, 2019-04-22:
I already tested gmsl camera driver on Xavier platform(driveworks1.5), however, I am not quite sure how to test this in docker because I did not have any customized image.
@gbiggs @jokla This is the thing that I want to discuss before I send any request. : )

amc-nu, 2019-04-23:
@twbabyduck We would appreciate if you can create the PR *without* docker support. The reason behind it, is that we cannot share docker images, scripts or content that violate the Nvidia license agreement.

twbabyduck, 2019-04-23:
@amc-nu 
Do you think you could guide me on how to build a customized docker image? 
I do have NVIDIA Xavier (Pegasus), so it is quite easy to test it.

sgermanserrano, 2019-04-23:
@twbabyduck have you had a look at the Dockerfiles inside the [_generic_](https://github.com/autowarefoundation/autoware/tree/master/docker/generic) folder? You should be able to reuse some of them

jokla, 2019-04-26:
Hi @twbabyduck, 

As @amc-nu suggested, could you create the PR without the docker support? Or create a public fork of Autoware with your modification? This would be helpful for the community. 

I didn't find any clear instructions on how to use docker on the PX2. There was a script but it is not working anymore.  #1928 

If you want to use the docker you should install Driveworks inside the docker or you should link the libraries from the host (easier but not very clean method).
See here for some references:
- [Installing Driveworks within a docker image](https://devtalk.nvidia.com/default/topic/1030912/faq/installing-driveworks-within-a-docker-image/
)
- [DriveInstall ](https://devtalk.nvidia.com/default/topic/1043762/driveworks/driveinstall-stuck-when-customizing-packages-installation-/?offset=2#5296679)

Nvidia-docker is not compatible with PX2. If you want to use the GPUs, you should also add the following instructions when you run the docker:
`--device=/dev/nvhost-ctrl --device=/dev/nvhost-ctrl-gpu --device=/dev/nvhost-prof-gpu --device=/dev/nvmap --device=/dev/nvhost-gpu --device=/dev/nvhost-as-gpu --device /dev/nvgpu-pci `

twbabyduck, 2019-04-30:
@jokla I still cannot access GPU(s) even using those flags to run Docker either on Pegasus or Xavier.
Does it work for you? If I want to install driveworks inside the docker, what exactly step should I follow?

jokla, 2019-04-30:
@twbabyduck yes, it is working for me. I have tested with the program called device query ([see](https://github.com/Technica-Corporation/Tegra-Docker#giving-docker-access-to-gpu) here for more info). Which command have you tried on the PX2 to check if the GPUs are accessible?

I have also a guide to run a docker with tensorflow with GPU support on the PX2. If you contact me on the [slack group ](https://autoware.herokuapp.com/) I can send you more information. Here probably we are a little bit off topic and we should discuss your pull request instead.

twbabyduck, 2019-05-02:
@jokla 
Thanks!
It is work now : )

closed 2020-05-29 by gbiggs