autowarefoundation/autoware #2125: Build error with catkin_make_release using CUDA

tneum, 2019-03-14:
Trying to install Autoware using the [Source-Build](https://github.com/CPFL/Autoware/wiki/Source-Build) instructions for Kinetic. Building with catkin_make using CUDA 9.0. The source installation works fine without CUDA. The docker installation works with CUDA.
 
# Bug
## Expected Behavior

Autoware building successfully using catkin_make and CUDA 9.0

## Actual Behavior

Build fails with following output:

[stderr.log](https://github.com/CPFL/Autoware/files/2965518/stderr.log)
[stdout.log](https://github.com/CPFL/Autoware/files/2965519/stdout.log)
[both.log](https://github.com/CPFL/Autoware/files/2965517/both.log)

```
[  3%] Linking CXX shared library /home/thomas/Autoware/ros/devel/lib/libndt_cpu.so
[  3%] Built target state_machine_lib
Scanning dependencies of target vg440_node
[  3%] Building CXX object sensing/drivers/imu/packages/memsic/CMakeFiles/vg440_node.dir/nodes/vg440/vg440_node.cpp.o
92 errors detected in the compilation of "/tmp/tmpxft_00003eac_00000000-6_NormalDistributionsTransform.cpp1.ii".
CMake Error at ndt_gpu_generated_NormalDistributionsTransform.cu.o.cmake:266 (message):
  Error generating file
  /home/thomas/Autoware/ros/build/computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/./ndt_gpu_generated_NormalDistributionsTransform.cu.o

computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/build.make:77: recipe for target 'computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o' failed
make[2]: *** [computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/src/ndt_gpu_generated_NormalDistributionsTransform.cu.o] Error 1
CMakeFiles/Makefile2:2181: recipe for target 'computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/all' failed
make[1]: *** [computing/perception/localization/lib/ndt_gpu/CMakeFiles/ndt_gpu.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  3%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/src/mip_sdk_user_functions.c.o
[  3%] Built target ndt_cpu
[  3%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip.c.o
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip.c: In function ‘mip_add_field’:
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip.c:189:3: warning: implicit declaration of function ‘memcpy’ [-Wimplicit-function-declaration]
   memcpy(field_data_ptr, field_data, data_size);
   ^
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip.c:189:3: warning: incompatible implicit declaration of built-in function ‘memcpy’
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip.c:189:3: note: include ‘<string.h>’ or provide a declaration of ‘memcpy’
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip.c: In function ‘mip_add_formatted_field’:
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip.c:250:2: warning: incompatible implicit declaration of built-in function ‘memcpy’
  memcpy(field_location, field, field_header_ptr->size);
  ^
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip.c:250:2: note: include ‘<string.h>’ or provide a declaration of ‘memcpy’
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip_sdk_base.c.o
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip_sdk_inteface.c.o
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip_sdk_3dm.c.o
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip_sdk_inteface.c: In function ‘__mip_interface_command_response_handler’:
/home/thomas/Autoware/ros/src/sensing/drivers/imu/packages/microstrain/MIPSDK/C/Library/Source/mip_sdk_inteface.c:755:49: warning: assignment from incompatible pointer type [-Wincompatible-pointer-types]
    device_interface->command_response_data      = field_header_ptr;
                                                 ^
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip_sdk_filter.c.o
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip_sdk_system.c.o
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip_sdk_ahrs.c.o
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/mip_sdk_gps.c.o
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Library/Source/ring_buffer.c.o
[  4%] Building C object sensing/drivers/imu/packages/microstrain/CMakeFiles/microstrain_driver.dir/MIPSDK/C/Utilities/Source/byteswap_utilities.c.o
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/laser_scan_converter/laser_scan_converter_node
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/image_processor/image_rotator
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/garmin/garmin_gps_18x_lvc
[  4%] Built target laser_scan_converter_node
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/image_processor/image_rectifier
[  4%] Built target image_rotator
[  4%] Built target garmin_gps_18x_lvc
[  4%] Linking CXX shared library /home/thomas/Autoware/ros/devel/lib/libmicrostrain_driver.so
[  4%] Built target microstrain_driver
[  4%] Built target image_rectifier
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/hokuyo/hokuyo_3d
[  4%] Built target hokuyo_3d
[  4%] Linking CXX shared library /home/thomas/Autoware/ros/devel/lib/libpcl_omp_registration.so
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/memsic_imu/vg440_node
[  4%] Built target vg440_node
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/point_cloud_converter/point_cloud_converter_node
[  4%] Linking CXX executable /home/thomas/Autoware/ros/devel/lib/map_tf_generator/map_tf_generator
[  4%] Built target pcl_omp_registration
[  4%] Built target point_cloud_converter_node
[  4%] Built target map_tf_generator
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
####
#### Running command: "make cmake_check_build_system" in "/home/thomas/Autoware/ros/build"
####
####
#### Running command: "make -j12 -l12" in "/home/thomas/Autoware/ros/build"
####
Invoking "make -j12 -l12" failed
```

## Steps to Reproduce the Problem

Short answer: Following the steps in [Source-Build](https://github.com/CPFL/Autoware/wiki/Source-Build).

Long answer:
```
  1. sudo apt-get update
  2. sudo apt-get install -y python-catkin-pkg python-rosdep python-wstool ros-$ROS_DISTRO-catkin libmosquitto-dev gksu
  3. cd $HOME
  4. git clone https://github.com/CPFL/Autoware.git --recurse-submodules
  5. cd ~/Autoware/ros/src
  6. catkin_init_workspace
  7. cd ../
  8. rosdep update
  9. rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
  10. ./catkin_make_release
```

## Specifications

  - Ubuntu version: 16.04 LTS
  - ROS version: Kinetic
  - Autoware branch: master (latest/1.10)
  - CUDA: 9.0 (also tried CUDA 8 and 10)
  - GPU: GTX 1060
  - Driver Version: 384.130
  - libpcl-dev: 1.7.2
  - gcc: 5.5

amc-nu, 2019-03-14:
I suspect this error is caused by your gcc version, please use `5.4.0` (the one shipped with Xenial).

tneum, 2019-03-14:
Thank you for the quick response, i will try that.

tneum, 2019-03-14:
Thank you, with 5.4 it is working fine!

amc-nu, 2019-03-14:
Great, thanks for reporting back.

closed 2019-03-14 by amc-nu