autowarefoundation/autoware #2128: points2vscan crashes

yuchenrao, 2019-03-14:
Fill-out only one section depending on whether you are reporting a bug or a new feature.
Do you have idea about why this happened, how to fix it or how to check the reason? Thanks a lot!

# Bug
## Expected Behavior
Points2Vscan keeps working can we can get /scan all the time

## Actual Behavior
Sometimes the points2vscan crashes and we cannot get the /scan data from it

## Steps to Reproduce the Problem

  1. Every time it crashes, we double check the velodyne points, and it has velodyne points.
  2. It happens in random situations

## Specifications

  - Ubuntu version:16.04
  - ROS version: kinetic
  - Autoware branch: master

# New Feature
Describe where the new feature request comes from. Add a high-level diagram of how the new feature should work.

gbiggs, 2019-03-14:
Thanks for using Autoware and for your bug report.

Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. We ask that you please follow our support guidelines when making a bug report: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

If you think that this may not be a bug but is instead a problem in your use of Autoware, then consider closing this bug report and opening a topic on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) requesting help. This may get you past your problem faster. Link to your new topic from this issue so we know where to continue the discussion.

wdxairforce, 2019-03-15:
@yuchenrao hi, do you just need the "scan" but not the "vscan" ? 
if yes, can you try to commit the following lines in mainwindow.cpp?
    for (int i = 0; i < BEAMNUM; i++)
    {
      double theta = i * density - PI;
      float* data;
      int* ring;

      data = (float*)(base + (2 * i) * msg.point_step);
      data[0] = beams[i] * cos(theta);
      data[1] = beams[i] * sin(theta);
      data[2] = virtualscan.minheights[i];
      data[4] = 255;
      ring = (int*)(data + 5 * sizeof(float));
      *ring = 0;

      data = (float*)(base + (2 * i + 1) * msg.point_step);
      data[0] = beams[i] * cos(theta);
      data[1] = beams[i] * sin(theta);
      data[2] = virtualscan.maxheights[i];
      data[4] = 255;
      ring = (int*)(data + 5 * sizeof(float));
      *ring = 1;
    }
    vsros->sendMessage(msg);
  }

yuchenrao, 2019-03-15:
> Thanks for using Autoware and for your bug report.
>
> Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. [...]

Thanks

yuchenrao, 2019-03-15:
> @yuchenrao hi, do you just need the "scan" but not the "vscan" ?
> if yes, can you try to commit the following lines in mainwindow.cpp?
> for (int i = 0; i < BEAMNUM; i++)
> {
> double theta = i * density - [...]

Thanks a lot for your reply. We just need laser scan data for obstacle detection. I will try to follow your suggestions and let you know how what result.

yuchenrao, 2019-03-15:
> Thanks for using Autoware and for your bug report.
>
> Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. [...]

Thanks a lot for your reply, I am not sure this is a bug, but will keep update the errors we meet.

amc-nu, 2019-03-15:
@yuchenrao `vscan`, `points2vscan`, `vscan2line` nodes are deprecated, and will be removed in the next release.
This node launches simultaneously three other [ones](https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/runtime_manager/scripts/vscan.launch). Even if `vscan` crashes, the other nodes will keep running. I suspect that is why you see some "output".

closed 2019-03-15 by amc-nu