autowarefoundation/autoware #2147: [Bug] Couldn’t find executable named rqt_lgsvl_simulator_configurator

MouriNaruto, 2019-03-24:
Fill-out only one section depending on whether you are reporting a bug or a new feature.

# Bug
## Expected Behavior
[image]

## Actual Behavior
[image]

## Steps to Reproduce the Problem
[image]
[image]

## Specifications
  - Ubuntu version: Ubuntu 16.04 in docker
  - ROS version: ROS Kinetic in docker
  - Autoware branch: master (https://github.com/CPFL/Autoware/commit/21e268d2beb20b325b2236186198b7ab2ffd4624)

I wonder to know how to solve. Thank you very much.

Mouri

gbiggs, 2019-03-24:
Thanks for using Autoware and for your bug report.

Your report does not give us enough information to reproduce the bug. If we cannot reproduce it, we will be unable to find the cause and fix it. We ask that you please follow our support guidelines when making a bug report: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

If you think that this may not be a bug but is instead a problem in your use of Autoware, then consider closing this bug report and opening a topic on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) requesting help. This may get you past your problem faster. Link to your new topic from this issue so we know where to continue the discussion.

ErijBNT, 2019-05-02:
did you manage to solve it?

MouriNaruto, 2019-05-02:
@ErijBNT I solve it by starting the simulator manually. But I still wait for the new autoware version.

AravindChandradoss, 2019-05-20:
@MouriNaruto  How did you solve? Can you explain the steps?

AravindChandradoss, 2019-05-22:
I am having this issue. Was this fixed? 

Thanks

gbiggs, 2019-05-23:
Thanks for using Autoware and for your bug report.

Your report does not give us enough information to reproduce the bug or confirm that it is actually a bug. If we cannot confirm it is a bug, we will be unable to assign it the correct priority. If we cannot reproduce it, we will be unable to find the cause and fix it. We ask that you please follow our support guidelines when making a bug report: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide there and in the issue template, and remember that a single screenshot does not describe expected behaviour nor actual behaviour.

In general, we prefer to resolve issues at ROS Answers first, and then make a bug report if the cause of the problem is identified as being a bug. This prevents the issue tracker from being filled with unconfirmed bugs and problems that are not actually bugs. It also helps our users get answers faster, both for the initial reporter because more people watch ROS Answers than watch the issue tracker, and for future users who have the same problem because ROS Answers is a searchable Q&A site. Please ask your question at [ROS Answers](https://answers.ros.org/questions/ask/?tags=autoware) requesting help.

MouriNaruto, 2019-05-23:
@gbiggs I have tried the newest commit in the master branch but still have this problem. I also tried on on other's PC and recreate new docker image. I don't know which more information can give.

amc-nu, 2019-05-23:
@hakuturu583 Can you please attend this problem.?

Seems you're missing an install command in the CMakeLists.txt for `scripts/rqt_lgsvl_simulator_configurator`.

amc-nu, 2019-05-23:
@MouriNaruto Please don't use ONLY screenshots. Read carefully the [support guidelines](https://github.com/autowarefoundation/autoware/wiki/Support-guidelines) and follow them. This will help to speed-up the identification and solution of the issue.

closed 2019-05-30 by gbiggs

AravindChandradoss, 2019-05-30:
**Quick Question:**

Was this commit merged with master branch? 

Because, I pulled master branch (Checked the CMakeLists.txt, it was edited (as expected [CMakeList.txt](https://github.com/autowarefoundation/autoware/commit/fcf5325cf3c3c2b5f867a82cf67e1360b34f012d) ) but when I re-compiled Autoware/ros/ I could not find the executable, rqt_lgsvl_simulator_configurator in /share/lgsvl_simulator_bridge.
And yes, got the same error.
`[rosrun] Couldn't find executable named rqt_lgsvl_simulator_configurator below /home/autoware/Autoware/ros/install/lgsvl_simulator_bridge/share/lgsvl_simulator_bridge`

Omega-Spinti, 2019-06-12:
so what is the fix? i am having the same issue. we can't start
 ```autoware@omega:~/Autoware/ros$ rosrun lgsvl_simulator_bridge rqt_lgsvl_simulator_configurator
[rosrun] Couldn't find executable named rqt_lgsvl_simulator_configurator below /home/autoware/Autoware/ros/install/lgsvl_simulator_bridge/share/lgsvl_simulator_bridge
 ```

this is what i get even when i try to run it manually

sriharshakunda, 2019-08-30:
@Omega-Spinti I am having same issue with 1.12.0, were you able to solve this issue?

pranavbajoria93, 2020-03-08:
@Omega-Spinti @amc-nu @MouriNaruto I have the same issue with Autoware 1.12.0 (docker). The launcher doesn't show up upon clicking the lgsvl simulator.
The cmakeLists.txt is located at `Autoware/src/autoware/simulation/lgsvl_simulator_bridge/` which has the correct paths like the expected commit. Is there a work around? How can it be configured manually?

manarhatem, 2020-07-18:
i am having the same issue can someone tell me the solution please?

> so what is the fix? i am having the same issue. we can't start
> 
> ```
> autoware@omega:~/Autoware/ros$ rosrun lgsvl_simulator_bridge rqt_lgsvl_simulator_configurator
> ```
> 
> ```
> [rosrun] Couldn't find executable named rqt_lgsvl_simulator_configurator below /home/autoware/Autoware/ros/install/lgsvl_simulator_bridge/share/lgsvl_simulator_bridge
> ```
> 
> this is what i get even when i try to run it manually
> 
> this is the list of folders:
> 
> ```
> autoware@omega:~/Autoware/ros$ ls install/lgsvl_simulator_bridge/share/lgsvl_simulator_bridge/
> cmake  config  hook  launch  media  package.bash  package.ps1  package.sh  package.xml  package.zsh  rviz
> ```

JWhitleyWork, 2020-07-19:
@manarhatem This is a very old thread and support-related questions should be asked on [ROS Answers](https://answers.ros.org/) per our [Support Guidelines](https://github.com/Autoware-AI/autoware.ai/wiki/Support-guidelines).

commit fcf5325c (2019-05-30): add install to the resources and plugin.xml (#2287) (Fixes #2147)