autowarefoundation/autoware #2152: Segmentation fault at velocity_set or unintended planning by asynchronous base_waypoints and closest_waypoint

kfunaoka, 2019-03-25:
## Bug report

### Status
Collecting information of this bug

### Required information:
- Operating system and version:
  - Ubuntu 16.04
- Autoware installation type:
  - From source
- Autoware version or commit hash
  - Ver.1.11.0
- ROS distribution and version:
  - Kinetic
- ROS installation type:
  - http://wiki.ros.org/kinetic/Installation/Ubuntu
- Package or library, if applicable:
  - Motion Planning

### Description of the bug
lane_select publishes base_waypoints and closest_waypoint. It's not guaranteed for astar_avoid to subscribe both of them at the same time. When base_waypoints is updated, closest_waypoint might point to the old base_waypoints in astar_avoid. If closest_waypoint is larger than the size of the old base_waypoints, astar_avoid publishes safety_waypoints with no entries. If velocity_set subscribes the safety_waypoints with no entries, velocity_set will die with segmentation fault.

### Steps to reproduce the bug
Collecting information

### Expected behavior
No segmentation fault and no unintended planning

### Actual behavior
Segmentation fault at velocity_set or unintended planning

### Screenshots
Collecting information

### Additional information
Collecting information

gbiggs, 2019-03-25:
I think we should avoid making incomplete bug reports. They will just clog up the issues with half-finished bug reports that we do not know how to reproduce.

Please post bug reports once the information to reproduce it has been collected so that developers can follow those steps and either confirm or reject the bug. If information on how to reproduce the bug is not available, then start a discussion on [discourse](https://discourse.ros.org/c/autoware) instead and post a bug report once you know how to reproduce it.

sgermanserrano, 2019-05-21:
Solved in #2272

closed 2019-05-21 by sgermanserrano