autowarefoundation/autoware #2156: Switch from submodule to using source repository for citysim

gbiggs, 2019-03-28:
## Feature request

### Description

Using git submodules makes it hard to manage branches and forks for the repository hosting them as well as for the submodule'd repositories. With the switch to using `vcs` to manage Autoware as a multi-repository project, we no longer need to rely on submodules. We should therefore remove the citysim submodule and replace it with an entry in the `.repos` file. Additionally we need to start using the upstream repository and contributing back to that rather than relying on our own fork.

### Implementation considerations

The current `autoware.repos` file contains an entry for citysim, but it points at the CPFL fork. This needs to be changed to point to the upstream repository and any changes made added to that via pull requests. The submodule entry in the `autoware` repository needs to be removed.

### Alternatives

None.

### Additional information

JWhitleyAStuff, 2019-05-20:
Unfortunately, https://bitbucket.org/osrf/citysim hasn't been maintained since 2017. We could probably get buy-in from OSRF for continued maintenance but this is just a note that it won't be as simple as creating the PRs and waiting for reviews.

gbiggs, 2019-05-23:
If it builds, then I think we can add it to the workspace. If it does not, then we will need to look at what depends on it and what the impact of dropping it will be.

Because it's simulation-related, we might be lucky and nothing depends on it directly for compilation.

gbiggs, 2019-06-17:
Fixed at GitLab.

closed 2019-06-17 by gbiggs