autowarefoundation/autoware #2158: Switch from submodule to using source repository for ouster

gbiggs, 2019-03-28:
## Feature request

### Description

Using git submodules makes it hard to manage branches and forks for the repository hosting them as well as for the submodule'd repositories. With the switch to using `vcs` to manage Autoware as a multi-repository project, we no longer need to rely on submodules. We should therefore remove the ouster submodule and replace it with an entry in the `.repos` file. Additionally we need to start using the upstream repository and contributing back to that rather than relying on our own fork. (If that makes sense; it was forked from an example.)

### Implementation considerations

The current `autoware.repos` file contains an entry for ouster, but it points at the CPFL fork. This needs to be changed to point to the upstream repository and any changes made added to that via pull requests. The submodule entry in the `autoware` repository needs to be removed.

### Alternatives

None.

### Additional information

amc-nu, 2019-03-28:
Same story as with #2159 
https://github.com/ros/rosdistro/pull/20556 and https://github.com/ros/rosdistro/pull/20555

gbiggs, 2019-03-28:
Same response, too. I'll remove it from the `autoware.repos` file and we will need to check the dependency flags are in place.

gbiggs, 2019-06-17:
Fixed at GitLab.

closed 2019-06-17 by gbiggs