autowarefoundation/autoware #2159: Switch from submodule to using source repository for robosense

gbiggs, 2019-03-28:
## Feature request

### Description

Using git submodules makes it hard to manage branches and forks for the repository hosting them as well as for the submodule'd repositories. With the switch to using `vcs` to manage Autoware as a multi-repository project, we no longer need to rely on submodules. We should therefore remove the robosense submodule and replace it with an entry in the `.repos` file. Additionally we need to start using the upstream repository and contributing back to that rather than relying on our own fork.

### Implementation considerations

The current `autoware.repos` file contains an entry for robosense, but it points at the CPFL fork. This needs to be changed to point to the upstream repository and any changes made added to that via pull requests. The submodule entry in the `autoware` repository needs to be removed.

### Alternatives

None.

### Additional information

amc-nu, 2019-03-28:
@gbiggs we tried to push to upstream, but haven't received any reply from the maintainers. 
https://github.com/RoboSense-LiDAR/ros_rslidar/pull/6

Since the code hasn't changed that much. I released the package to the ROS farm, it will be ready in the next sync.  https://github.com/ros/rosdistro/pull/20693 and https://github.com/ros/rosdistro/pull/20692

gbiggs, 2019-03-28:
Either way, we can get rid of the submodule, so I'm mostly happy.

We will need to check that the necessary dependency tags are in the packages that need it so the binary gets pulled in by rosdep.

sgermanserrano, 2019-03-28:
Once the binary is available it needs to be added [here](https://github.com/autowarefoundation/autoware/blob/master/docker/generic/dependencies) so that we have a clear list of ROS dependencies

gbiggs, 2019-06-17:
Fixed at GitLab.

closed 2019-06-17 by gbiggs