autowarefoundation/autoware #2163: CI checks for the autoware.repos file

gbiggs, 2019-03-29:
## Feature request

### Description

The `autoware.repos` file will define what is in a particular release of Autoware. Because it will involve pointing at various repositories and tags or even specific commits in those repositories, we need to make sure the information contained in the file works (e.g. is proper YAML, points at existing commits, etc.).

### Implementation considerations

A simple Python script could do the job.

### Alternatives

### Additional information

https://github.com/autowarefoundation/autoware/pull/2161#pullrequestreview-220335122

JWhitleyAStuff, 2019-03-29:
This might help (just added): https://github.com/dirk-thomas/vcstool/pull/90

JWhitleyAStuff, 2019-03-29:
https://github.com/dirk-thomas/vcstool/pull/90

esteve, 2019-04-01:
Given that we'll use `vcs` to check out the repositories, we can just make `vcs import` the first step of the build process and it'll fail if the YAML file is incorrect or if the repositories can't be accessed.

JWhitleyAStuff, 2019-04-01:
@esteve The problem with using the import command as a test is that, if it succeeds, it will download whatever the endpoint is (repo, tar file, etc.). This can take a very long time, especially with repos with as much history as the Autoware main repo. Hopefully my change to vcstool above will solve this problem. The only downside to my method is that it can not validate the existence of the contents of the version tag. It can, however, contact the repository URI and validate that the endpoint is of the correct type (git, hg, svn, zip, or tar).

JWhitleyAStuff, 2019-05-20:
Still waiting on review from @dirk-thomas.

JWhitleyAStuff, 2019-06-25:
Merged. Can be closed.

closed 2019-06-25 by gbiggs