autowarefoundation/autoware #2167: Autoware simulation get "stuck" after some time

vingtfranc, 2019-03-29:
## Bug report

### Required information:

- Operating system and version:
  - Ubuntu 16.04
- Autoware installation type:
  - Compiled from release. No docker
- Autoware version or commit hash
  - 1.10.0
- ROS distribution and version:
  - kinetic
- ROS installation type:
  - apt-get

### Description of the bug
I'm running moriyama rosbag with a few modules on my system (lidar cluster detection and ndt_matching)
In Rviz I can see that the demo runs fine: I can see clusters being detected around the car, and I can see the car driving on the vector map as expected. However after some amount of time a weird "bug" occurs: 
- TF msg are not received anymore by Rviz. I can see new camera images, new lidar point cloud, and new detected clusters, but nothing will move. The gps/map/car positions are not updated anymore.
- I cannot pause the simulation anymore, when I hit pause, the simulation keeps going (even though the button is pushed). The only way I can get the control back is to restart the simulation

When I look at the ros logs, I see that ndt_matching and lidar_cluster_detect are still running fine after the bug occured. Actually I don't see anything strange.

Do you have an idea of what could be the cause?

vingtfranc, 2019-03-29:
I have added a second machine to my setup in order to play the rosbag
On the first machine I start autoware normally
On the second machine I also start autoware but without roscore (need to edit ./run script), and I load the rosbag here instead of the first machine. 
I don't have any issue with this setup.

So this might be related to the hardware I'm using or maybe a bug in ROS?

Note: I first tried to simply run the `rosbag play -r 1 --clock <bag>` from the second computer but some /tf msgs were not published, so I guess Autoware does more than just run the rosbag play when the user start the simulation. Is that correct?

gbiggs, 2019-03-31:
Thanks for using Autoware and for your bug report. However, I think there is a good chance that this is a problem in your use of Autoware rather than a bug. Please make a post on Discourse following our support guidelines: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide. This will get you help faster, and if we confirm that it is a bug then we can re-open this issue with the new information.

closed 2019-03-31 by gbiggs